eyal0/OctoPrint-PrintTimeGenius #133: [Feature Request] Push remaining time to LCD with M73 command

Mortikhan, 2019-06-12:
OctoPrint Version:1.3.11

Some printers support the M73 command to display the estimated time remaining. It would be nice if we could push the PTG estimated time remaining to the LCD display as it is way more accurate. Perhaps anytime an embedded m73 code is sent, PTG could insert/send a new m73 with its more accurate estimate.

eyal0, 2019-06-12:
I think that there are already other plugins that can do this.  For example: https://plugins.octoprint.org/plugins/m73progress/

You should be able to use this together with PTG.  Let me know if it doesn't work

Sciguy429, 2019-06-15:
The M73progress plugin dose seem to function alongside this one, however it dose not pick up the percentage generated by PTG. It seems to just divide the two sizes displayed above the progress bar.

eyal0, 2019-06-19:
That's usually how it goes on the octoprint display, too. The webpage shows the percentage of file done. Having two different percentages seemed too confusing so I never did it.

If it were worth doing, it would make sense for the M73 plug-ins, of which there are a few,  to do it. If you're interested, I'd reach out to those authors.

Mortikhan, 2019-06-21:
M73 Progress plugin does mostly work. same issues as posted above regarding percentage complete.
Thanks. This at least gets me a better ETiC on my LCD screen based on the PTG values.

eyal0, 2020-02-04:
This one does progress based on time: https://plugins.octoprint.org/plugins/ProgressBasedOnTime/

According to the author, it works together with PTG.  Great!

closed 2020-02-04 by eyal0