eyal0/OctoPrint-PrintTimeGenius #134: Error reading file list on startup

jneilliii, 2019-06-16:
OctoPrint Version: 1.3.11

PrintTimeGenius Version (if you know): 1.3.1

What did you try:

What happenned: Upon initial connection/startup of the UI an error is thrown in developer console indicating an issue reading the file list. My assumption is the file list isn't initialized/bound yet prior to being called by your plugin. Here's a screenshot.

[image]

What did you expect to happen: no errors on startup.

eyal0, 2019-07-03:
Does it happen each time?

jneilliii, 2019-07-13:
It doesn't seem to happen every time.  I need to try and see a common occurrence when it does. Maybe the first load after a full restart/reload? Only reason I even saw the error was during development which for me typically involves a bunch of install/reinstall of plugins forcing restart/reloads.

jneilliii, 2019-09-14:
So I can confirm, it happens every time you open the browser after initial startup.  The error is pointing back to [here](https://github.com/eyal0/OctoPrint-PrintTimeGenius/blob/518409e0ad19d91014f9aae1e058567c57096d95/octoprint_PrintTimeGenius/static/js/PrintTimeGenius.js#L62).  I think this is probably occurring because of the asynchronous nature of the file list and therefore on initial startup items/self.filesViewModel.allItems() is null. Maybe converting the function call to recurseFiles to utilize ko.utils.arrayFilter methods instead may get around this.  An example can be found [here](http://www.knockmeout.net/2011/04/utility-functions-in-knockoutjs.html) in the `Filtering an array` section.

eyal0, 2019-09-15:
Sounds like you'd know about it more than I!  Would you be willing to make a PR about it?

jneilliii, 2019-09-16:
After looking at the code a little more and reviewing the fileListViewModel.allItems binding in OctoPrint it was actually way simpler than changing the filtering logic. I've submitted a pull request of the change that resolves the issue in my testing.

eyal0, 2019-09-16:
Cool.  Great solution.  In your testing it works and that's enough for me.  Thanks!

PR #152 (2019-09-16): filesViewModel.allItems() binding issue
Resolves #134 by making it's containing div in settings use an `if` data-binding for the file list in analyzer section.
closed 2019-09-16 by eyal0