eyal0/OctoPrint-PrintTimeGenius #136: Manual analyze not working at all. (no errors, no log output)

ChJaeger, 2019-06-22:
**OctoPrint Version:** 
1.3.11

**PrintTimeGenius Version (if you know):**
1.3.1

**What did you try:**
Manually analyze a file from the settings.

**What happenned:**
TL;DR: Nothing. No output in log. No yellow star on the file.

More Infos: When I installed the plugin I got yellow stars on some of my files that already where on the server. I could't make out why those files where chosen (some where already successfully printed, some where not printed yet, All files used PrusaSlicer). 
I then tried to manually analyze some of the files that didn't get the star, but nothing happend. No error messages and no output at all in the log file. Same if I try to analyze a file that already has a star. 

If the file is on the top level or in a folder does not make a difference.

If I upload new files to the server, the files analyze automatically (blinking star) und I get a valid log message. (The times are within minutes of what prusaSlicer tells me, so I guess it working). 

My settings are in the screenshot. Logfile is included

**What did you expect to happen:**
File beeing analyzed., when clicking the button.

**If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).**

[image]
[plugin_PrintTimeGenius_engine-19.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3316739/plugin_PrintTimeGenius_engine-19.log)

eyal0, 2019-06-24:
Strange.  I don't see anything wrong in the log file.  Maybe there's a javascript error?  You can see the javascript console by pressing `F12` on most browsers.  Perhaps look at the console while trying to analyze?

ChJaeger, 2019-06-30:
Yes it's really strange. Especially that its only this one button that doesn't work. Everything else (including other plugins) is working fine.
I did some more testing and tried  from different machines running MacOS and Linux. I still get the same behaviour no matter what operating system (and browser).

There is no related output to the javascript console.

I tried uninstalling and reinstalling the plugin which didnt help either. 

I then logged onto my pi via ssh and ran the command from the settings in bash
`
"/home/pi/OctoPrint/venv/bin/python2" "/home/pi/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "{gcode}"`
Obvioulsy exchanging the {gcode} with one of my not yet analyzed files and then I get a valid output to the console, but still no star on my file and no entry to the PTG-Log.

The Attachement is the console output when running the command in bash.
[console_output.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3342777/console_output.log)

I have Octoprint installed on a stock Raspian, I'm not usin OctoPI. Can this be an issue?

eyal0, 2019-07-03:
Hmm.  It should not matter that you did not use OctoPi.

That command that you typed does the analysis but it does not save it into OctoPrint, which is why you didn't get a star.  But the output looks correct.

When you click that analyze button, there is nothing added to the octoprint plugin logs?  You can find the log file for the plugin in the ~/.octoprint/logs directory.  Can you check both the PTG log and the octoprint log?  I am surprised...

ChJaeger, 2019-07-08:
Hi. I just checked. I dont get any entry in the octoprint log or the PTG log when pressing the analyze button. I only get entries for new uploads (automatically).

Knifa, 2019-07-08:
Seeing the same issue myself.

Knifa, 2019-07-08:
As far as I can tell, the Analyze button literally does nothing. No network calls, no output in the JS console, etc.

closed 2019-07-09 by eyal0

eyal0, 2019-07-09:
I think that I fixed it in the latest version.  Can you try to install this: https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/master.zip (from the normal OctoPrint install dialog, you can put a URL).

New features:
* You can select multiple items to analyze.
* They are sorted so that the unanalyzed are at the top and then alphabetically.
* I fixed the button, it was broken.

eyal0, 2019-07-09:
After you comment on the behavior, I'll make a new version.

ChJaeger, 2019-07-09:
I installed the new version. It works perfectly now ! Thank you very much for all the effort.

eyal0, 2019-07-09:
How do you like the new look, with the list and multiselect? Ok? If so, I'll push the new version very soon.

ChJaeger, 2019-07-09:
I like it.👍 Having the un-analyzed files at the top makes sense and the feedback with the hourglass is also nice so you can see that something is happening.

PR #138 (2019-07-09): Fix quoting of optionsCaption in settings
This fixes #136