eyal0/OctoPrint-PrintTimeGenius #137: Printer Monitor progress bar

Jeff-Haas, 2019-06-25:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version:
1.3.11
PrintTimeGenius Version (if you know):
1.3.1
What did you try:
Printing progress bar shown on Printer Monitor does not match what's shown on Octoprint.  Please see photo.

Printer Monitor is here:
https://github.com/Qrome/printer-monitor

What happened:
Progress bar shown on Printer Monitor is behind the one displayed in Octoprint.

What did you expect to happen:
Both progress bars should match.

In addition, author of Printer Monitor says:
"The monitor displays the value from progress->completion value in the /api/job call."
This makes me think that the progress bar percentage is possibly not reported via the API the same as shown on screen (but this is speculation).

[image]

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

eyal0, 2019-06-27:
Ah, I see.  I fixed this in the webpage before but I forgot that I did it: https://github.com/eyal0/OctoPrint-PrintTimeGenius/commit/cd41885348e0a27005c73ba67d2af4fdce28dbdd

That makes it so the percentage is of time, not file size.  You can see that 2:18 is a little less than half 2:44 and 45% is a little less than half.

But that is a cosmetic fix, it only fixes the webpage.  It doesn't repair the reporting that your print-monitor is using.  I think that the correct solution is for you to reach out to the creators of your device and have them fix it.  It shouldn't be too hard for them to do it!

eyal0, 2019-07-09:
As discussed in the other repo, we can't and won't fix this here.

closed 2019-07-09 by eyal0