eyal0/OctoPrint-PrintTimeGenius #141: PTG Stopped working as of latest release

Mortikhan, 2019-07-10:
OctoPrint Version:1.3.11
PrintTimeGenius Version (if you know):1.3.2

What did you try:Updated to latest version of PTG and PTG stopped functioning.

What happened: No Stars on file list(blinking or solid), No processing of new files. Analyzers section of PTG is blank(no commands just blank line). Files to analyze is also blank.

eyal0, 2019-07-10:
Hmm.  Can you open the java console and reload OctoPrint and see what it says?  To open the java console, press F12.

Also, can you check that PTG is installed?  It should be in the OctoPrint plugin manager.

Mortikhan, 2019-07-10:
PTG is installed on BOTH instances of Octoprint and BOTH instances are exhibiting the same behavior.
No idea what I'm looking for in the Java console.
Here are screenshots https://imgur.com/S3fSHEE

Mortikhan, 2019-07-11:
snapshot of the java console https://imgur.com/H8lQqIO

Mortikhan, 2019-07-11:
I disabled all 3rd party plugins except PrintTimeGenius and rebooted the pi.
No change in behavior or settings.
The "Reset Analyzers to Defaults" also appears to do nothing.

Mortikhan, 2019-07-12:
I tried to copy and paste the default analyzers into the PTG plugin.
I am only able to add one line. Pressing on the + button does not add addtional space/line for me to add more entries.

Mortikhan, 2019-07-12:
I think there is a conflict between PTG and one of the other plugins I am running.
I reinstalled from a fresh octopi image, and PTG has no analyzers and does not display stars on the main screen

Here is a list of the plugins I installed
CancelObjects 0.3.3
DeleteAfterPrint 1.3.1
Enclosure Plugin 4.12
M73 ETA Override 1.0.1
Octolapse 0.3.4
Octoslack 1.9.0
PrintTimeGenius 1.3.2
Tab Order 0.5.3
Themeify 1.20

PR #144 (2019-07-12): Use _.has() in case gcodeAnalysis is missing
Fixes #141 
Fixes #142 
Fixes #143
closed 2019-07-12 by eyal0