eyal0/OctoPrint-PrintTimeGenius #143: File list no longer updating after upgrading to 1.3.2

Flogue, 2019-07-12:
OctoPrint Version: 1.3.11

PrintTimeGenius Version (if you know): 1.3.2

What did you try: Manage gcode files from UI.

What happenned: The file list never gets updated (on delete/add, etc.). Only a page reload update the file list. This behavior stops after uninstalling PTG. I did not observe this behavior in 1.3.1.

What did you expect to happen: The file list to be dynamically updated.

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

I see the flowing error in the console while loading the octoprint server, every time:

```Could not bind view model PrintTimeGeniusViewModel to target #settings_plugin_PrintTimeGenius : TypeError: Unable to process binding "options: function(){return FileList }"
Message: Cannot read property 'progress' of undefined
    at eval (eval at parseBindingsString (http://octopi.local/static/webassets/packed_libs.js?ea9fbc9a:11374:90), <anonymous>:4:52)
    at f (http://octopi.local/static/webassets/packed_libs.js?ea9fbc9a:11399:330)
    at http://octopi.local/static/webassets/packed_libs.js?ea9fbc9a:11401:395
    at a.B.i (http://octopi.local/static/webassets/packed_libs.js?ea9fbc9a:11424:24)
    at Function.Pc (http://octopi.local/static/webassets/packed_libs.js?ea9fbc9a:11357:355)
    at Function.Qc (http://octopi.local/static/webassets/packed_libs.js?ea9fbc9a:11357:133)
    at Function.aa (http://octopi.local/static/webassets/packed_libs.js?ea9fbc9a:11356:482)
    at Object.a.m.a.B (http://octopi.local/static/webassets/packed_libs.js?ea9fbc9a:11355:39)
    at b (http://octopi.local/static/webassets/packed_libs.js?ea9fbc9a:11424:5)
    at Object.a.a.Bb (http://octopi.local/static/webassets/packed_libs.js?ea9fbc9a:11426:80)```

closed 2019-07-12 by eyal0

eyal0, 2019-07-12:
The log that you provided was very helpful in helping me understand the problem.  Thanks!

I have released version 1.3.3

Flogue, 2019-07-12:
Awesome!

PR #144 (2019-07-12): Use _.has() in case gcodeAnalysis is missing
Fixes #141 
Fixes #142 
Fixes #143