eyal0/OctoPrint-PrintTimeGenius #146: I have the same problem since a while, today updated to PrintTImeGenius 1.3.4, but the error remains, print time is not calculated. I'm using Octoprint 1.3.11. Here is the last log:

eyal0, 2019-07-27:
I have the same problem since a while, today updated to PrintTImeGenius 1.3.4, but the error remains, print time is not calculated. I'm using Octoprint 1.3.11. Here is the last log:
````
2019-07-21 18:33:17,649 Not running built-in analysis.
2019-07-21 18:33:17,654 Disabled: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/E12_PLA_Intex_Planeholder_mod.gcode"
2019-07-21 18:33:17,655 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/E12_PLA_Intex_Planeholder_mod.gcode" "M200 D1.75
M200 D0
M92 X80.00 Y80.00 Z400.00 E216.00
M203 X300.00 Y300.00 Z5.00 E0.00
M201 X800 Y800 Z100 E10000
M204 P800.00 R3000.00 T800.00
M205 B20000 E0.00 S0.00 T0.00 X8.00 Y8.00 Z0.30"
2019-07-21 18:33:24,463 Failed to run '"/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/E12_PLA_Intex_Planeholder_mod.gcode" "M200 D1.75
M200 D0
M92 X80.00 Y80.00 Z400.00 E216.00
M203 X300.00 Y300.00 Z5.00 E0.00
M201 X800 Y800 Z100 E10000
M204 P800.00 R3000.00 T800.00
M205 B20000 E0.00 S0.00 T0.00 X8.00 Y8.00 Z0.30"'
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 318, in _do_analysis
    raise Exception(results_err)
Exception: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/E12_PLA_Intex_Planeholder_mod.gcode" "M200 D1.75
M200 D0
M92 X80.00 Y80.00 Z400.00 E216.00
M203 X300.00 Y300.00 Z5.00 E0.00
M201 X800 Y800 Z100 E10000
M204 P800.00 R3000.00 T800.00
M205 B20000 E0.00 S0.00 T0.00 X8.00 Y8.00 Z0.30"
..Processed 232707 Gcodes and 18 Mcodes. 222262 blocks
Total time: inf
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 90, in <module>
    main()
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 63, in main
    result.update(json.loads(line))
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
````
The gcode was created with Simplify3D. Any suggestions, since about a month ago, everything worked fine... ?
Thank you!

_Originally posted by @vogtitec in https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/145#issuecomment-513578169_

eyal0, 2019-07-27:
@vogtitec Please send your gcode file.  Thanks.

vogtitec, 2019-07-28:
OK, thanks, here is the file:

[E12_PLA_Intex_Planeholder_mod.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3438976/E12_PLA_Intex_Planeholder_mod.zip)

eyal0, 2019-07-29:
In the command that you are using, there is M203 E0.  That means that the maximum extruder feedrate is set to zero, which is no good.

Can you type M503 on your printer in the console to see all the current settings?  Check if the max extruder rate is actually set to 0.

Also, after you type M503, PTG might see this and learn the new extruder rate.  Try to analyze the file again after you type M503 and it might work better.

vogtitec, 2019-07-30:
Oh, I am quite astonished! That indeed is strange. Since I didn't change anything, the reason might be a recent update of Simplify3D. I'll check this!
Thank you very much for your inspection!

vogtitec, 2019-08-13:
Today I corrected the max feedrate setting for E in my firmware, and everything works fine now. Thank you for your help!

closed 2019-08-13 by eyal0