eyal0/OctoPrint-PrintTimeGenius #147: [Feature Request] Keep track of estimates vs actual print times

ewortzman, 2019-08-10:
OctoPrint Version: 1.3.11

It would be a nice feature if PrintTimeGenius had a panel, either on the main view or just in settings, where it keeps track of the estimated print time that was calculated prior to the print starting and compares it to the actual print time when the print is complete.  At a minimum, this could give the user the ability to manually adjust the estimates to get a better idea of how long the actual print is going to take.  It could also in theory be used to adjust the estimation function in the future to further improve the accuracy of the estimates.

eyal0, 2019-08-11:
It already does this.  Can you look in the advanced panel of the PTG options and see if the table there meets your needs?

ewortzman, 2019-08-12:
I don't see anything there that looks quite like what I'm going for.  There's a description of how to create a graph to compare the PTG estimates to Octoprint's built-in estimates, and there is also a way to download a log, which seems to contain information about the estimate, but in a non-structured way that is difficult to parse.  It also doesn't appear to store the actual time the print took, either.

What I'm looking for is a simple table with just a few columns:

- File name
- PTG estimated print time
- actual print time (after the job is completed)

Obviously if the print fails or is cancelled, it wouldn't be included.

eyal0, 2019-08-13:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/How-does-it-work%3F#how-printtimegenius-compensates

Can you take a screenshot of your PTG options screen after expanding the advanced options? There should be a table there.

eyal0, 2019-08-13:
Ever see something like this?

https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/127#issuecomment-505147493

ewortzman, 2019-08-30:
Sorry for the delay.  Yes I do see that, although the table is empty.  Do I need to turn anything on to get it to work?  I've had several successful prints with no interruptions or anything and the table doesn't have any values in it.

ewortzman, 2019-09-08:
@eyal0 Sorry to ping you. Just wondering if you saw this.

eyal0, 2019-09-09:
Sorry, I was busy.

I don't have a printer available to me so I tried it in simulation and it seems to be working fine for me.  After I printing, I was able to see the compensation values in the settings.  Unfortunately, I think that I had to restart OctoPrint to see them there.  I think that they'll take effect even if I don't restart but to see them there, I need to restart.

They're also in the `~/.octoprint/config.yaml` file and called "print_history".  So you should be able to see it there, too.

Maybe your system is unable to write to that config.yaml file, like it's write-protected or something?

The config changes are only written/read when the system quits/starts.  That's to prevent lots of reading and writing on the file, which can be kind of slow.

Try the restarting like I said and also check the config.yaml file to see if anything is getting written there.  Thanks!

ewortzman, 2019-09-09:
Thanks for the response and sorry to bother you.

I see the values in config.yaml, but I'm not seeing them loaded into the table, even after restarting octoprint. I'm not sure if the compensation values are working, but I'm getting some really inaccurate estimates (I just had a print estimated at 28 hours that finished in 22, for instance). Is there anything I can do to verify that the compensation values are being picked up properly?

eyal0, 2019-09-09:
The log will show compensation when a file is analyzed. When compensation values change, files are **not** reanalyzed. So you might need to analyze then again. Files that are uploaded are always analyzed with the latest compensation**at the time of the upload**.

eyal0, 2019-09-15:
Maybe there is some error in the log file that you see?  Or an error in your browser that appears in the browser's console?

ewortzman, 2019-09-18:
Hmm.  I did find this in the browser console.  Looks like it could be related.  Didn't find anything in the log file, though maybe I didn't look int he right place.

Also, it looks like everything has the appropriate permissions in `~/.octoprint` so I don't think that's an issue.

```
Could not bind view model PrintTimeGeniusViewModel to target #settings_plugin_PrintTimeGenius : PrintTimeGeniusViewModel/self.recurseFiles@http://192.168.1.98/static/webassets/packed_plugins.js?ba7736f4:634:32
PrintTimeGeniusViewModel/self.FileList<@http://192.168.1.98/static/webassets/packed_plugins.js?ba7736f4:646:27
Pc@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11357:355
Qc@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11357:133
aa@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11356:482
sa@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11358:332
X@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11342:441
uc@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11356:171
Y@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11314:246
oc@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11345:325
e@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11353:419
c@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11325:1
update@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11400:70
m/</<@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11378:286
Pc@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11357:355
Qc@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11357:133
aa@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11356:482
a.B@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11355:39
m/<@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11378:273
q@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11317:244
m@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11377:318
k@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11375:387
g@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11375:278
k@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11375:440
g@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11375:278
k@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11375:440
g@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11375:278
k@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11375:440
g@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11375:278
k@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11375:440
g@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11375:278
k@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11375:440
a.Rb@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11381:413
bindViewModels/</<@http://192.168.1.98/static/webassets/packed_core.js?8c230bbf:15939:36
Pn@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11208:530
ur/<@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11229:66
bindViewModels/<@http://192.168.1.98/static/webassets/packed_core.js?8c230bbf:15909:27
Pn@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11208:530
ur/<@http://192.168.1.98/static/webassets/packed_libs.js?ea9fbc9a:11229:66
bindViewModels@http://192.168.1.98/static/webassets/packed_core.js?8c230bbf:15876:15
```

eyal0, 2019-09-18:
Yeah, that's not good! The SettingsViewModel is how you watch config.yaml from PTG.

It works for most people. Maybe there is a conflict with a different plugin? Does PTG work if other plugins are disabled?

eyal0, 2019-11-14:
Still waiting on an answer about potential conflicts with other plugins...  I can't figure out why you're having an issue, sorry.

Re-open this if it's still an issue or if you have an ideas what I should look into.  Maybe there's something special about your rig?

On September 24 I made a new version of PTG with a much better marlin-calc.  You might get better estimates now.

closed 2019-11-14 by eyal0