eyal0/OctoPrint-PrintTimeGenius #148: Filament Volume Calculation wrong

Peter-J, 2019-08-15:
OctoPrint Version: 1.3.11
PrintTimeGenius Version (if you know): 1.3.5

Somehow the Volume-Calculation defaults to a Diameter of 3mm resulting in a volume of 11.97 cm^3 instead of using the filament-diameter 1.75mm end ending up with ~4cm^3
There is no M200-G-Code in the file. The file is generated by Slic3r-PE and contains the line
> ; filament_diameter = 1.75"

Logfile: (volume numbers highlighted for ease of use)
> 2019-08-15 02:21:17,519 Not running built-in analysis.
> 2019-08-15 02:21:17,523 Running: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Handy-Zusatz-9.gcode"
> 2019-08-15 02:21:19,224 Sarge output: 
> 2019-08-15 02:21:19,226 Result: {"estimatedPrintTime": 546.0, "filament": {"tool0": {"volume": **4.0**, "length": 1673.7}}}

> 2019-08-15 02:21:19,244 Merged result: {'analysisPending': True, u'estimatedPrintTime': 546.0, u'filament': {u'tool0': {u'volume': **4.0**, u'length': 1673.7}}}
> 2019-08-15 02:21:26,948 Running: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Handy-Zusatz-9.gcode" ""
> 2019-08-15 02:21:31,238 Sarge output: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/Handy-Zusatz-9.gcode"
Processed 23729 Gcodes and 598 Mcodes. 22940 blocks
Total time: 515.087241

> 2019-08-15 02:21:31,289 Result: {"printingArea": {"maxZ": 15.0, "maxX": 189.723, "maxY": 157.955, "minX": 60.0, "minY": -3.0, "minZ": 0.0}, "dimensions": {"width": 129.723, "depth": 160.955, "height": 15.0}, "firstFilament": 0.00205662507681973, "progress": [[0, 515.0872405737281], [0.0019380011147794, 509.90673827057793], [0.00205662507681973, 503.90546435863513], [0.11490945990367163, 443.9026044454124], [0.2373522559990853, 383.7661437962697], [0.35499721305149423, 323.73777013445954], [0.46873472537838184, 263.5072105773089], [0.5829567379839643, 203.3572832326031], [0.6976175163286599, 143.34913096715212], [0.8211536537609513, 83.32623451231672], [0.9432906001229115, 23.18118897548851], [0.9880146922208407, 3.6598042993706486], [0.9883348339979133, 0.0], [1, 0]], "estimatedPrintTime": 515.0872405737281, "lastFilament": 0.9880146922208407, "filament": {"tool0": {"volume": **11.97**6870752569882, "length": 1694.3805600000053}}}

> 2019-08-15 02:21:31,320 Merged result: {u'printingArea': {u'maxZ': 15.0, u'maxX': 189.723, u'maxY': 157.955, u'minX': 60.0, u'minY': -3.0, u'minZ': 0.0}, u'dimensions': {u'width': 129.723, u'depth': 160.955, u'height': 15.0}, 'analysisPending': True, u'firstFilament': 0.00205662507681973, u'progress': [[0, 515.0872405737281], [0.0019380011147794, 509.90673827057793], [0.00205662507681973, 503.90546435863513], [0.11490945990367163, 443.9026044454124], [0.2373522559990853, 383.7661437962697], [0.35499721305149423, 323.73777013445954], [0.46873472537838184, 263.5072105773089], [0.5829567379839643, 203.3572832326031], [0.6976175163286599, 143.34913096715212], [0.8211536537609513, 83.32623451231672], [0.9432906001229115, 23.18118897548851], [0.9880146922208407, 3.6598042993706486], [0.9883348339979133, 0.0], [1, 0]], u'estimatedPrintTime': 515.0872405737281, u'lastFilament': 0.9880146922208407, u'filament': {u'tool0': {u'volume': **11.97**6870752569882, u'length': 1694.3805600000053}}}
> 2019-08-15 02:21:35,637 Running: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Handy-Zusatz-9.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining
> 2019-08-15 02:21:39,215 Sarge output: 
> 2019-08-15 02:21:39,222 Result: {"estimatedPrintTime": 546.0}

> 2019-08-15 02:21:39,225 Merged result: {u'printingArea': {u'maxZ': 15.0, u'maxX': 189.723, u'maxY': 157.955, u'minX': 60.0, u'minY': -3.0, u'minZ': 0.0}, u'dimensions': {u'width': 129.723, u'depth': 160.955, u'height': 15.0}, 'analysisPending': True, u'firstFilament': 0.00205662507681973, u'progress': [[0, 515.0872405737281], [0.0019380011147794, 509.90673827057793], [0.00205662507681973, 503.90546435863513], [0.11490945990367163, 443.9026044454124], [0.2373522559990853, 383.7661437962697], [0.35499721305149423, 323.73777013445954], [0.46873472537838184, 263.5072105773089], [0.5829567379839643, 203.3572832326031], [0.6976175163286599, 143.34913096715212], [0.8211536537609513, 83.32623451231672], [0.9432906001229115, 23.18118897548851], [0.9880146922208407, 3.6598042993706486], [0.9883348339979133, 0.0], [1, 0]], u'estimatedPrintTime': 546.0, u'lastFilament': 0.9880146922208407, u'filament': {u'tool0': {u'volume': **11.97**6870752569882, u'length': 1694.3805600000053}}}
> 2019-08-15 02:21:43,319 Recent heat-up times in seconds: 119.368591422, 53.440549704, 141.472933796, 137.648406885, 135.029677202
> 2019-08-15 02:21:43,321 Average heat-up: 117.392031802 seconds
> 2019-08-15 02:21:43,323 Recent cool-down times in seconds: 46.877660666, 100.540385489, 29.252333485, 39.86716148, 19.753212774
> 2019-08-15 02:21:43,324 Average cool-down: 47.2581507788 seconds
> 2019-08-15 02:21:43,325 Time spent printing, actual vs predicted: 18011.6261455/13155.1651416, 4904.74145827/3455.61780943, 3290.52835123/3305.39740057, 3342.81238457/3368.58938664, 30529.3230501/30540.3657253
> 2019-08-15 02:21:43,326 Average scaling factor: 1.15520168998
> 2019-08-15 02:21:43,328 Compensated result: {'analysisFirstFilamentPrintTime': 42.09453564136487, u'printingArea': {u'maxZ': 15.0, u'maxX': 189.723, u'maxY': 157.955, u'minX': 60.0, u'minY': -3.0, u'minZ': 0.0}, u'dimensions': {u'width': 129.723, u'depth': 160.955, u'height': 15.0}, 'analysisPending': False, 'analysisPrintTime': 546.0, 'analysisLastFilamentPrintTime': 542.3401957006294, u'firstFilament': 0.00205662507681973, u'progress': [[0, 742.5348144881366], [0.00205662507681973, 625.1427826863364], [0.11490945990367163, 555.8273775107191], [0.2373522559990853, 486.35763653918383], [0.35499721305149423, 417.01275783808035], [0.46873472537838184, 347.43431364898873], [0.5829567379839643, 277.94901592798055], [0.6976175163286599, 208.62749701812507], [0.8211536537609513, 139.28894559577526], [0.9432906001229115, 69.80928734747742], [0.9880146922208407, 47.25815077880179], [1, 0]], u'estimatedPrintTime': 742.5348144881366, u'lastFilament': 0.9880146922208407, u'filament': {u'tool0': {u'volume': **11.97**6870752569882, u'length': 1694.3805600000053}}}

eyal0, 2019-08-15:
Hmm, yes, that's not right.  `marlin-calc` doesn't look at those comments, it only looks at things like M200.  However, I would have expected an M200 to be in your command line!  That is isn't there is odd.  PTG should have picked it up from whenever you last ran M503.

Could you please try to run M503 and then go into the options and do another analysis of a file?  I expect that instead of:

> 2019-08-15 02:21:26,948 Running: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Handy-Zusatz-9.gcode" **""**

in your log, you should see something like this:

> 2019-08-15 02:21:26,948 Running: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Handy-Zusatz-9.gcode" **"M200 something here
M201 Xwhatever Yblahblahblah Zetc
M202 more stuff
M205 even more stuff"**

PTG should automatically grab that stuff out of the M503 command that you run if you do M503 while you aren't printing.  Please try and let me know if it helps.

Peter-J, 2019-08-15:
My Prusa Mk2s seems to be M200 agnostic

output of M503:
> Send: M503
> Recv: echo:Steps per unit:
> Recv: echo:  M92 X100.00 Y100.00 Z400.00 E161.30
> Recv: echo:Maximum feedrates (mm/s):
> Recv: echo:  M203 X500.00 Y500.00 Z12.00 E120.00
> Recv: echo:Maximum Acceleration (mm/s2):
> Recv: echo:  M201 X9000 Y9000 Z500 E10000
> Recv: echo:Acceleration: S=acceleration, T=retract acceleration
> Recv: echo:  M204 S1000.00 T1500.00
> Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
> Recv: echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.20 E2.50
> Recv: echo:Home offset (mm):
> Recv: echo:  M206 X0.00 Y0.00 Z0.00
> Recv: echo:PID settings:
> Recv: echo:   M301 P19.21 I1.55 D59.40
> Recv: echo:PID heatbed settings:
> Recv: echo:   M304 P74.37 I3.17 D435.61
> Recv: echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)
> Recv: echo:   M207 S3.00 F2700.00 Z0.00
> Recv: echo:Recover: S=Extra length (mm) F:Speed (mm/m)
> Recv: echo:   M208 S0.00 F480.00
> Recv: echo:Auto-Retract: S=0 to disable, 1 to interpret extrude-only moves as retracts or recoveries
> Recv: echo:   M209 S0
> Recv: echo:Filament settings: Disabled
> Recv: ok

And the logfile:
> 2019-08-15 03:06:04,378 Not running built-in analysis.
> 2019-08-15 03:06:04,383 Running: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Handy-Zusatz-9.gcode"
> 2019-08-15 03:06:05,949 Sarge output: 
> 2019-08-15 03:06:05,958 Result: {"estimatedPrintTime": 546.0, "filament": {"tool0": {"volume": 4.0, "length": 1673.7}}}

> 2019-08-15 03:06:05,997 Merged result: {'analysisPending': True, u'estimatedPrintTime': 546.0, u'filament': {u'tool0': {u'volume': 4.0, u'length': 1673.7}}}
> 2019-08-15 03:06:12,008 Running: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Handy-Zusatz-9.gcode" "**M92 X100.00 Y100.00 Z400.00 E161.30
M203 X500.00 Y500.00 Z12.00 E120.00
M201 X9000 Y9000 Z500 E10000
M204 S1000.00 T1500.00
M205 B20000 E2.50 S0.00 T0.00 X10.00 Y10.00 Z0.20
M207 F2700.00 S3.00 Z0.00
M208 F480.00 S0.00**"
> 2019-08-15 03:06:16,719 Sarge output: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/Handy-Zusatz-9.gcode" "**M92 X100.00 Y100.00 Z400.00 E161.30
M203 X500.00 Y500.00 Z12.00 E120.00
M201 X9000 Y9000 Z500 E10000
M204 S1000.00 T1500.00
M205 B20000 E2.50 S0.00 T0.00 X10.00 Y10.00 Z0.20
M207 F2700.00 S3.00 Z0.00
M208 F480.00 S0.00**"
Processed 23729 Gcodes and 605 Mcodes. 22984 blocks
Total time: 512.984252

> 2019-08-15 03:06:16,724 Result: {"printingArea": {"maxZ": 15.0, "maxX": 189.723, "maxY": 157.955, "minX": 60.0, "minY": -3.0, "minZ": 0.0}, "dimensions": {"width": 129.723, "depth": 160.955, "height": 15.0}, "firstFilament": 0.00205662507681973, "progress": [[0, 512.9842521694503], [0.0019380011147794, 507.8035394364347], [0.00205662507681973, 501.8020582036485], [0.11495233603452958, 441.69122346435694], [0.23881004444825565, 381.6879636098404], [0.3563692492389487, 321.6414045480062], [0.4700210093041204, 261.53357287883983], [0.5847146593491404, 201.40027670955186], [0.7009189784047221, 141.3714713101361], [0.825762837828181, 81.36987223804954], [0.9476596778573368, 21.35929510073703], [0.9880146922208407, 3.6598210998095055], [0.9883348339979133, 0.0], [1, 0]], "estimatedPrintTime": 512.9842521694503, "lastFilament": 0.9880146922208407, "filament": {"tool0": {"volume": **11.97**6870752569882, "length": 1694.3805600000053}}}

> 2019-08-15 03:06:16,734 Merged result: {u'printingArea': {u'maxZ': 15.0, u'maxX': 189.723, u'maxY': 157.955, u'minX': 60.0, u'minY': -3.0, u'minZ': 0.0}, u'dimensions': {u'width': 129.723, u'depth': 160.955, u'height': 15.0}, 'analysisPending': True, u'firstFilament': 0.00205662507681973, u'progress': [[0, 512.9842521694503], [0.0019380011147794, 507.8035394364347], [0.00205662507681973, 501.8020582036485], [0.11495233603452958, 441.69122346435694], [0.23881004444825565, 381.6879636098404], [0.3563692492389487, 321.6414045480062], [0.4700210093041204, 261.53357287883983], [0.5847146593491404, 201.40027670955186], [0.7009189784047221, 141.3714713101361], [0.825762837828181, 81.36987223804954], [0.9476596778573368, 21.35929510073703], [0.9880146922208407, 3.6598210998095055], [0.9883348339979133, 0.0], [1, 0]], u'estimatedPrintTime': 512.9842521694503, u'lastFilament': 0.9880146922208407, u'filament': {u'tool0': {u'volume': **11.97**6870752569882, u'length': 1694.3805600000053}}}
> 2019-08-15 03:06:19,439 Running: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Handy-Zusatz-9.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining
> 2019-08-15 03:06:24,031 Sarge output: 
> 2019-08-15 03:06:24,032 Result: {"estimatedPrintTime": 546.0}

> 2019-08-15 03:06:24,034 Merged result: {u'printingArea': {u'maxZ': 15.0, u'maxX': 189.723, u'maxY': 157.955, u'minX': 60.0, u'minY': -3.0, u'minZ': 0.0}, u'dimensions': {u'width': 129.723, u'depth': 160.955, u'height': 15.0}, 'analysisPending': True, u'firstFilament': 0.00205662507681973, u'progress': [[0, 51 [... cut ...] 1204, 345.15434463640617], [0.5847146593491404, 275.688259277347], [0.7009189784047221, 206.34288183223197], [0.825762837828181, 137.02893318242587], [0.9476596778573368, 67.70461305649798], [0.9880146922208407, 47.25815077880179], [1, 0]], u'estimatedPrintTime': 740.1049367352783, u'lastFilament': 0.9880146922208407, u'filament': {u'tool0': {u'volume': **11.97**6870752569882, u'length': 1694.3805600000053}}}

Should i try ... manually adding M200 D1.75 to the Marlin Firmware simulation command?

Peter-J, 2019-08-15:
OK i just typed M200 D1.75 in the console and re-analyzed the file ... now it's getting funny

> 2019-08-15 03:17:44,759 Sarge output: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/Handy-Zusatz-9.gcode" "M92 X100.00 Y100.00 Z400.00 E161.30
M203 X500.00 Y500.00 Z12.00 E120.00
M201 X9000 Y9000 Z500 E10000
M204 S1000.00 T1500.00
M205 B20000 E2.50 S0.00 T0.00 X10.00 Y10.00 Z0.20
M207 F2700.00 S3.00 Z0.00
M208 F480.00 S0.00
**M200 D1.75**"
Processed 23729 Gcodes and 606 Mcodes. 22856 blocks
Total time: 507.232101

> 2019-08-15 03:17:44,760 Result: {"printingArea": {"maxZ": 15.0, "maxX": 189.723, "maxY": 157.955, "minX": 60.0, "minY": -3.0, "minZ": 0.0}, "dimensions": {"width": 129.723, "depth": 160.955, "height": 15.0}, "firstFilament": 0.00205662507681973, "progress": [[0, 507.2321011867482], [0.0019380011147794, 502.05138845373256], [0.00205662507681973, 496.0499309767191], [0.11727050550958282, 436.01752408107603], [0.24250025011076334, 375.97521829132063], [0.3598850919693007, 315.97199018012566], [0.47529191499092455, 255.86517614331353], [0.5906601494947763, 195.74987004163404], [0.7072732209978705, 135.7005392311856], [0.8333247581071618, 75.65135548932767], [0.9586631222398491, 15.500984997797048], [0.9880146922208407, 3.648009835726839], [0.9883348339979133, 0.0], [1, 0]], "estimatedPrintTime": 507.2321011867482, "lastFilament": 0.9880146922208407, "filament": {"tool0": {"volume": **1.69**43805164870127, "length": 704.4415403710235}}}

eyal0, 2019-08-15:
I expected that in your M503 output it would say:

```
M200 D1.75
M200 D0
```

both of those.  The first line enables volumentric extrusion with 1.75mm diameter.  The second line turns off the volumetric extrusion.

Look at the marlin-calc command line in here:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/146

It has both `M200 D1.75` and then `M200 D0`

I am surprised that there is no `M200` in your `M503` output at all.  Usually those are in there.  That's weird.  You are using marlin firmware?

Anyway, type `M200 D1.75` and then `M200 D0` and try again please.  Thanks.

Peter-J, 2019-08-15:
Yey it works. So typing
M200 D1.75
M200 D0
helped.

My printer is a Prusa-i3-MK2s ... so i guess a Prusa-flavour of marlin?

> 2019-08-15 03:33:24,968 Sarge output: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/Handy-Zusatz-9.gcode" "**M200 D1.75
M200 D0**
M92 X100.00 Y100.00 Z400.00 E161.30
M203 X500.00 Y500.00 Z12.00 E120.00
M201 X9000 Y9000 Z500 E10000
M204 S1000.00 T1500.00
M205 B20000 E2.50 S0.00 T0.00 X10.00 Y10.00 Z0.20
M207 F2700.00 S3.00 Z0.00
M208 F480.00 S0.00"
Processed 23729 Gcodes and 607 Mcodes. 22984 blocks
Total time: 512.984252

> 2019-08-15 03:33:24,970 Result: {"printingArea": {"maxZ": 15.0, "maxX": 189.723, "maxY": 157.955, "minX": 60.0, "minY": -3.0, "minZ": 0.0}, "dimensions": {"width": 129.723, "depth": 160.955, "height": 15.0}, "firstFilament": 0.00205662507681973, "progress": [[0, 512.9842521694503], [0.0019380011147794, 507.8035394364347], [0.00205662507681973, 501.8020582036485], [0.11495233603452958, 441.69122346435694], [0.23881004444825565, 381.6879636098404], [0.3563692492389487, 321.6414045480062], [0.4700210093041204, 261.53357287883983], [0.5847146593491404, 201.40027670955186], [0.7009189784047221, 141.3714713101361], [0.825762837828181, 81.36987223804954], [0.9476596778573368, 21.35929510073703], [0.9880146922208407, 3.6598210998095055], [0.9883348339979133, 0.0], [1, 0]], "estimatedPrintTime": 512.9842521694503, "lastFilament": 0.9880146922208407, "filament": {"tool0": {"volume": **4.07**546296441614, "length": **1694**.3805600000053}}}

Maybe you could add somewhere in the installation instructions or in the wiki that one should check if "M503" gives M200 values and if not one should enter the filament-diameter manually with those two M200-commands?

Peter-J, 2019-08-15:
Ok, this fix doesn't survive a printer & octoprint reboot ... i should be able to find a straight forward solution from here on myself.

eyal0, 2019-08-15:
Yeah it should!  The settings aren't saved continuously because it would tax the system.  Instead, they are only saved when OctoPrint is shutdown.  Did you shutdown OctoPrint properly by going into the settings in the top right corner and selecting reboot or shutdown?

Peter-J, 2019-08-15:
Clean shutdown ... cranced logging for PrintTimeGenius to "debug".
Still amnesia

[octoprint.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3506794/octoprint.log)
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3506795/plugin_PrintTimeGenius_engine.log)

Should saving the config be visible in the log?

eyal0, 2019-11-13:
The saving should be in the logs now.  Try again?

eyal0, 2020-01-05:
In #167 I fixed the save of the configuration.  Maybe this will fix your problem that the configuration is saved across reboots now?

eyal0, 2020-01-05:
Also, if you don't have M200 in the M503 output, you might have M200 in the gcode file itself?  PTG can use either.