eyal0/OctoPrint-PrintTimeGenius #149: Feature request

Thisismydigitalself, 2019-08-24:
Hi Eyal, 

Any chance for your genius plugin (Thank u!) to command octoprint to open this pull down window once PTG is done doing its thingy?.. 

[image]

eyal0, 2019-08-31:
I don't think that this is something that could or should be part of PTG.  I think that this should be a whole different plugin, if that's something that is needed.

Why do you need it?

Thisismydigitalself, 2019-09-04:
Hi,
When your Plugin finishes its calculation and yellow star stops blinking the pull down menu stays closed and you don't see the results. you need to open the window to see those. the upper octoprint information above is updated only after starting the print job. i don't see a reason why your plug should not present the results when they are ready.

This is when calculation is done:
[image]

This is when i manually click on this arrow down/open file tab and see the results:
[image]