eyal0/OctoPrint-PrintTimeGenius #150: [Feature request] Show estimated time until filament change

Tronnic, 2019-08-27:
OctoPrint Version: OctoPrint 1.3.11 running on OctoPi 0.16.0
PrintTimeGenius Version: 1.3.5

Hello there, would it be possible to look for M600 (Change Filament) in the GCode and give an estimated time until the next filament swap? 

This would be wonderful. Thank you :-) 

Best regards,
Frank.

eyal0, 2019-08-30:
Huh, that is a good idea.  Let me think this through.

Would it make sense to do it as part of this plugin or in a different plugin?  It might make sense in a different plugin.  I need to check if there is a generalized way to ask OctoPrint "What is the time left until hitting XX% percentage of the file?"  Where the XX% is the time until the M600 in the file.  I think that could be not too hard to do.

I will look into that and also check to see if there are other plugins that already do this.

eyal0, 2019-08-30:
I didn't find anything like that in my search here: https://plugins.octoprint.org/search/

Tronnic, 2019-08-30:
I didn't find anything either. Would love to have this integrated into PTG. :-)

Tronnic, 2019-10-01:
Any news on this? :-)

eyal0, 2019-10-04:
Still just thinking about it.  Lately I'm trying to think about whether it ought to be part of this plugin or a new one.

yoyo42, 2019-12-13:
I think this needs to be a PTG thing, to figure out closely enough where the filaments change(s) are going to happen you need a good analysis of the gcode, not just a percentage of the way through it. From a UI point of view I'd either have an extra line showing 'time to next filament change' (which makes it work the same for any number of changes) or else a complete PTG tab.

I don't know what multi-filament printers use to switch filament - this would only be required for 'time to next _manual_ filament change'.

eyal0, 2020-01-10:
I'm considering working on this.  I don't own a multi-filament printer.  I don't even own any printers at all.  If I did this work, would someone be willing to test it?

yoyo42, 2020-02-02:
I'd be happy to test it. As I said before it's only really useful on single filament printers anyway as it's all about when you need to switch the colour of that single filament.

eyal0, 2020-02-04:
What needs to happen is for octoprint's update progress to send not just time until completion but also time until next filament change.  The plugin to do this would monkey patch the original function that computes progress with a new one that calls the original and then adds to it more info for when the next filament change is.

The plugin should send to octoprint the regular expressions matching filament change (or whatever) and also the text to display, below the existing ETA display.  The script should also know what kind of test to display.  So maybe the regex is "M600" and the text to display is "Time to filament: {timeToFilament}".  Something like that.

There will be a javascript piece to modify the website so that it'll display that right below the usual progress line.

eyal0, 2020-02-05:
@Tronnic @yoyo42 I've started work on this.  Nothing ready yet but maybe within a week.

In the meanwhile, can you provide sample gcode?  https://github.com/eyal0/OctoPrint-TimeToFilament/issues/1

Thanks.

eyal0, 2020-02-06:
This is a "rough draft".  It shows the seconds until the next M600.  It'll only work if you're already using PTG.  I hope that it doesn't interfere with other plugins.  Let me know if it's at all working or accurate on your setup.

You can install it from within OctoPrint, I think!  I think that you just need this link:

https://github.com/eyal0/OctoPrint-TimeToFilament/archive/master.zip

jr-air, 2020-05-20:
Quick questions:
1) Is the TimetoFilament change merged into the TimeGenius or needed separate?
2) Is the M600 gcode only discovered during upload? (asking because I often insert this manually after uploading then manually inserting with the gcode editor)

eyal0, 2020-05-20:
1. It's not merged in.  They are two separate plugins.  You must have PTG installed to use TimeToFilament.
2. The gcode is scanned while the program is running so you can modify the gcode after you upload it.

Good questions!  Let me know if you find it useful.

jr-air, 2020-05-20:
Happened to have a print job with Change Filament and Octoprint alerted me to this update. :)
Let me know what additional info you would like (after this job completes).

The one thing I noticed is the time. It says "seconds" when think it's hours.

[image]

eyal0, 2020-05-20:
Yeah, that's just a cosmetic issue.  I'll fix that.  It's actually showing you hours:minutes:seconds.

You can fix it yourself, too, in the advanced settings.  You just need to remove the word "seconds" from the "format" of the two display lines.  You can do it even while printing and it will take effect immediately.

KenLucke, 2020-05-21:
> If you want OctoPrint to report the time until the next filament change or next layer, let me know #150

YES, please, and I'll test.

eyal0, 2020-05-21:
@KenLucke Just install PTG v2.2.2 or later if you haven't already and then install this from within OctoPrint (you can put the url in there):

https://github.com/eyal0/OctoPrint-TimeToFilament/archive/master.zip

It should work.  Let me know how it works and also let me know about the accuracy of the results, please.

KenLucke, 2020-05-21:
> @KenLucke Just install PTG v2.2.2 or later if you haven't already and then install this from within OctoPrint (you can put the url in there): [...]

Already done.  I found then url upthread.    I have a filament change coming in the next file I am printing, so should be able to test then.

Off-topic for the thread, but any chance you and the guy working on the ArcWelder plugin (reducing jitter, stalls, blobs, and zits due to serial buffer inadequacies by converting straight segments into G2/G3 arcs) could get together and solve the massive time errors that occur when using PTG and ArcWelder?

jr-air, 2020-05-21:
An update. On my print that included a filament change, the time estimate was optimistic. 

The time to reach the M600 gcode was 5:30 hrs. Above the estimate displayed was originally  ~2:24 hrs.

(I do wish my printer was as fast as estimated!)

eyal0, 2020-05-21:
That's not great!  Hmm...  What about the total print time estimate?  Was it also very wrong or was it okay?

The estimate gets recalculated as the print continues.  Did it converge toward the right answer or did it just snap from way off to zero?

I don't have a printer at all so I'm unable to testing things.

KenLucke, 2020-05-23:
Any chance of adding M0 (pause) to time estimates as well as the M600 (Change Filament)?

eyal0, 2020-05-27:
Sure, though you could add it yourself, too.  Just modify the advanced settings in the plugin.  Change the `M600` to `M0`.  You can also test that it's working by modifying the sample gcode at the bottom and look at the sample output in the settings page.

The change takes effect immediately so you can do it while you print, too.

rben13, 2020-05-28:
I'm using a Prusa Mk3S, and PrusaSlicer , which uses `M600` for Color Change and `M601` for Pause. I used Advanced Settings to set up the pause correctly. But I wasn't getting any lines on the OctoPrint display that showed the time to next layer, time to color change, or time to pause. I disabled the time to layer and time to pause, but I'm still not getting any output on the Octoprint display.

I don't think I'm seeing any output from the plugin. Perhaps I'm looking in the wrong place?

eyal0, 2020-05-28:
Hmm, maybe I need to think of a better way to debug this thing because so many people are seeing errors.

Can you show me your configuration to make sure that there are no errors?

rben13, 2020-05-29:
I am now seeing a line that shows filament change in 00:00:00. I think in at least one case, I was looking at a file that did not have a filament change, so that was my mistake. Now the problem seems to be that the time is not showing.

The actual text I see in the terminal is:
`Send: N{some number} M600*42`

The set up is the default. I have tried both with Time to Next Layer enabled and disabled, it doesn't seem to work at all, but I haven't verified how the layers show up in the GCode on the Prusa, yet.

eyal0, 2020-05-29:
All that looks fine. I'll try to make a new version that logs more info. We'll see if that's helpful

eyal0, 2020-05-29:
Okay, I made a new version.  You can install it in the same way:

https://github.com/eyal0/OctoPrint-TimeToFilament/archive/master.zip

After you install it, go into settings and look for "logging" and then add "TimeToFilament" for debug logging.  There might be two "TimeToFilaments" in the list, just add them both, no harm.

This will cause a log file to be written in the .octoprint/logs directory, with the same TimeToFilament.  The log shows what TimeToFilament is sending to your web browser.  It will have rows like this while you print:

```
2020-05-28 20:44:39,930 sending: {"completion": 10.902862006301488, "filepos": 3146204, "printTime": 297, "printTimeLeft": 5391, "printTimeLeftOrigin": "genius", "TimeToFilament": {"^M600": {"timeLeft": 5374.7848675227315, "groups": [], "group": "M600", "groupdict": {}, "matchPos": 3315448, "searchPos": 5687}, "^; layer (\\d+)": {"timeLeft": 5372.014810196676, "groups": ["42"], "group": "; layer 42", "groupdict": {}, "matchPos": 3343149, "searchPos": 2956343}}}
```

That one means that the printer is currently 10.9% of the way through a file, printing currently byte 3146204 of the file.  It has been printing so far for 297 seconds and we think that there are 5391 more seconds to go.

TimeToFilament found an `^M600` in the file, and we expect to reach that line with 5374 seconds more of printing left.  Since we're at 5391 left right now, that means that the time until the filament change is (5391-5374) 17 seconds from now.  We also found a layer file, with time left 5372.  We're at 5391 left so the time until the next layer starts printing will be 5391-5372, so 19 seconds from now.  The group found was "42" so it's layer 42.  We started our search for it at position 2956343 in the file and we found that layer line at position 3343149.

If you send me one of yours, I can read it.  It can also help if you can send the output of the PTG analysis, which is found in the PTG log.  If you send both of those and even the gcode file, too, then I can maybe piece it together.  I hope!

Thanks for your patience.

rben13, 2020-05-29:
I will try to get that information together for you and will try the new version as soon as the current print finishes. One thing that might be a clue is that when there are two filament changes, the second one does show up with a time.

eyal0, 2020-05-29:
That *is* a good clue. I have caching in the system so that if we've previously found a filament change, we don't go searching for it again. However, maybe I did it wrong. I did have some bugs around that before.

I hope that the debug logs will provide a good answer. Ideally you can send me three things about a print:

1. The gcode printed.
2. The log of the analysis of the file from PTG logs.
3. The log of TTF with debug enabled.

I think that with those three pieces, I could sort out what's going on.

rben13, 2020-05-31:
My wife tells me that the last version works for us. :) We will try some more.

eyal0, 2020-07-16:
This is complete, I'm just waiting for it to get on the plugins repo.  For now, you can install this zip file i nthe OctoPrint Plugin Manager in the settings:

https://github.com/eyal0/OctoPrint-TimeToFilament/archive/master.zip

closed 2020-07-16 by eyal0