eyal0/OctoPrint-PrintTimeGenius #151: PTG 1.3.5 PI3B High CPU Uasge

Thisismydigitalself, 2019-09-04:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version:
1.3.10 and/or 1.3.11
PrintTimeGenius Version (if you know):
1.3.5
What did you try:
loading  OctoPrint page while printer is in active Print job
What happenned:
CPU usage goes up to 112% on Pi3B.

What did you expect to happen:
this spike in cpu usage effects printer. print head stutters due to USB streaming issues which lasts as long as CPU is over 100%. when i disable PTG pluging in octoprint plugin manager from loading, CPU is under 40% when loading OP page on remote host.

**CPU usage when PTG is enabled:**
---------------------------------------
3672 pi        20   0  194596  60848   7888 S   3.0  6.8   1:25.76 octoprint
 3672 pi        20   0  194596  60868   7888 S   3.0  6.8   1:25.85 octoprint
 3736 root      20   0    1900   1264   1124 S   0.3  0.1   0:00.01 octoprint
 3750 pi        20   0   44240  21812   6636 R  79.8  2.4   0:02.41 octoprint
 3750 pi        20   0   83852  33216   7416 R 100.0  3.7   0:05.43 octoprint
 3750 pi        20   0  103000  42944   7416 R 111.3  4.8   0:08.79 octoprint
 3750 pi        20   0  128264  42712   7476 S  27.8  4.8   0:09.63 octoprint
 3750 pi        20   0  156940  47388   7472 S  60.8  5.3   0:11.46 octoprint
 3750 pi        20   0  180680  46956   7492 R 111.3  5.2   0:14.82 octoprint
 3750 pi        20   0  182728  48780   7492 R 113.9  5.4   0:18.26 octoprint
 3750 pi        20   0  183648  49872   7492 R 113.6  5.6   0:21.69 octoprint
 3750 pi        20   0  185604  51972   7492 R 114.2  5.8   0:25.14 octoprint
 3750 pi        20   0  185984  52424   7492 S 113.6  5.8   0:28.56 octoprint
 3750 pi        20   0  185916  51976   7492 R 113.6  5.8   0:31.99 octoprint
 3750 pi        20   0  186940  52156   7492 S 109.6  5.8   0:35.30 octoprint
 3750 pi        20   0  187552  53420   7492 R 109.6  6.0   0:38.61 octoprint
 3750 pi        20   0  192636  57312   7712 R 106.6  6.4   0:41.83 octoprint
 3750 pi        20   0  196380  61648   7712 R 107.0  6.9   0:45.05 octoprint
 3750 pi        20   0  195744  61668   7712 R 115.2  6.9   0:48.53 octoprint
 **3750 pi        20   0  199856  65184   7712 S 114.9  7.3   0:52.00 octoprint**
 3750 pi        20   0  194360  59916   7712 S 113.9  6.7   0:55.44 octoprint
 3750 pi        20   0  195052  60796   7712 S 108.6  6.8   0:58.71 octoprint
 3750 pi        20   0  194360  60088   7712 R 103.6  6.7   1:01.84 octoprint
 3750 pi        20   0  194360  60088   7712 S  22.5  6.7   1:02.52 octoprint
 3750 pi        20   0  194360  60088   7712 S   2.3  6.7   1:02.59 octoprint

**CPU usage when PTG is disabled (top during page load):**
-------------------------------------------------------------
3750 pi        20   0  194360  60088   7712 S   0.7  6.7   1:02.61 octoprint
 3750 pi        20   0  194360  60088   7712 S   0.3  6.7   1:02.62 octoprint
 3750 pi        20   0  194360  60088   7712 S   2.6  6.7   1:02.70 octoprint
 **3750 pi        20   0  194360  60088   7712 S  33.3  6.7   1:03.71 octoprint**
 3750 pi        20   0  194360  60088   7712 S   8.6  6.7   1:03.97 octoprint
 3750 pi        20   0  194360  60088   7712 S   0.7  6.7   1:03.99 octoprint
 3750 pi        20   0  194360  60088   7712 S   2.0  6.7   1:04.05 octoprint
 3750 pi        20   0  194360  60088   7712 S   0.7  6.7   1:04.07 octoprint

eyal0, 2019-09-04:
Interesting.  I wonder what it's doing at that time...  this happens when you reload the page?  Like, you press F5 while the printer is printing?

Thisismydigitalself, 2019-09-04:
cntl+R in chrome for reload page.
When PTG is enabled, each time i reload the page cpu usage spikes as high as 115%.
this does not happen now when I have disabled PTG. I really like PTG and use it all the time but i noticed too many streaming issues that I had to find out what goes on. i did not know it is PTG. i disabled all plugins and enabled each at a time. PTG looks to be the culprit.

Thisismydigitalself, 2019-09-04:
Doubled and tripled checked. PTG, when enabled, pushes cpu to as high as 115% when page reloads. when disabled load does not go over 70% max. this is now without an active print job. tomorrow will test while printer is printing and hopefully this will not causes any issues while loading/reloading page.

Thisismydigitalself, 2019-09-05:
This morning checked refresh/reload page during active print job. this time when PTG is disabled.
I am both happy and sad to declare it was the PTC plugin which when enabled greatly effects pi3B performances. now when PTC is disabled i can refresh/reload/open new sessions of OP on different hosts and printer keeps printing without stuttering. i am sure more tests should be performed but for sure there is a real noticeable differences between when PTG is enabled and disabled.

FYI.

Thisismydigitalself, 2019-09-09:
Today I bumped up USB communication speed between OP and Smoothieboard from 115K to 250K.
OP/Printer wise all is well and working fast and nice. so i decided to try and reload PTG and see if, for any reason, this will help and prevent communication error but sadly it did not.
PTG is defiantly pushes cpu usage very high when browser refreshes / reload OP page.

:(

eyal0, 2019-09-11:
So weird.  Is there anything in the log to suggest what it is doing during the high CPU usage.  Maybe it's trying to recalculate all the estimated print times?  If that were the case, you'd see a lot of gold stars being worked on, added, etc.  But if that were the case, I'd expect `top` to report calls to marlin-calc.

I'm not sure how to figure out what it is doing.  There is an option in the settings to use OctoPrint's built-in print time calculation.  That is known to be very slow and it isn't necessary when PTG is installed.  Maybe you have that setting?

Thisismydigitalself, 2019-09-12:
Are you referring to this?
If so, disabling this plug did not help.
PTG during refresh and for some reason is doing something (or causing something else to) overload the CPU. 
[image]

eyal0, 2019-09-30:
I just fixed a bug about 100% CPU but it was for marlin-calc.  #159 . I wonder what OctoPrint is so busy doing? I'm not sure how to figure it out. If you kill OctoPrint when it gets into that state, maybe it would print out a backtrace showing where it was stuck?

Thisismydigitalself, 2019-10-08:
Hi Eyal, I am not sure I know what "marlin-calc" is all about. 
I have two printers, each got its own octopi. the printer i notice this issue (hard to miss..)  is the one that runs marlin. the other one runs Smoothieware. i need to check PTC with this other printer and report back, but this reported issue is for my Pi pushing an 8bit board running Marlin. I am not sure if my pi cpu load is way over 100% due to waiting for an OK reply from printer, or PTC is working hard regardless. I do not see the fact that this board is 8bit but it was worth mentioning.

Thisismydigitalself, 2020-08-14:
Hi Eyal, I hope this comment finds you happy and smiling.
A year a go i stopped (unwillingly) using PTG due to high CPU usage on my Pi3.
I love PTG and would like to start using it again but first i must know it will not effect my prints due to intermediate gcode streaming issues due to cpu bottleneck. 

Any news regarding optimization?
Cheers.