eyal0/OctoPrint-PrintTimeGenius #153: Linear Advance in marlin-calc

gpouger, 2019-09-16:
Hi,

It seems that Linear Advance is not enabled in your build of marlin-calc, this makes the estimates quite a lot too optimistic for me (a small print estimated to 6:30 min by PTG actually took 12:15 min). As a result I get some crazy compensation values...

Could you enabled it in a future release? (#define LIN_ADVANCE in Configuration_adv.h).
If you enable it with a default value of 0 (#define LIN_ADVANCE_K 0) it should not change the estimates for people not using it.

Thanks a lot for your work!

eyal0, 2019-09-16:
Can you send me an example gcode file with the settings used to run marlin-calc that lead to LIN_ADVANCE making a big difference?  I would like to see how much effect LIN_ADVANCE is having on the timing.

For the arguments to marlin-calc, you can check the PTG log file

gpouger, 2019-09-16:
Here are the parameters:
```
2019-09-16` 08:45:33,547 Running: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/single_wall_cylinder_layer.gcode" "M200 D1.75
M200 D0
M92 X80.00 Y80.00 Z1600.00 E415.00
M203 X300.00 Y300.00 Z7.00 E70.00
M201 X3000 Y3000 Z100 E10000
M204 P1000.00 R2000.00 T2500.00
M205 B20000 E5.00 S0.00 T0.00 X7.00 Y7.00 Z0.20
M207 F3600.00 S3.00 W13.00 Z0.00
M208 F3600.00 S0.00 W0.00"
```
The gcode is attached.
[single wall cylinder layer.gcode.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3616463/single.wall.cylinder.layer.gcode.txt)

eyal0, 2019-09-16:
I tried it with both LIN_ADVANCE and without and it had no effect on timing.  Are you sure that LIN_ADVANCE is supposed to have an effect on the time to print?

I'm reading through the marlin code but it's not clear to me that it should have an effect.  I read this, too: http://marlinfw.org/docs/features/lin_advance.html

Is it possible that the estimates that you got were way off because of the heat-up and cool down times of the printer?  The compensation tables should show this information and it's also in the PTG log.

gpouger, 2019-09-16:
Depending on the jerk and acceleration settings, Linear Advance may change the X/Y movement accelerations to match the modified filament flowing speed.
I'm sure that if I disable Linear Advance my prints go much faster (but are much less clean).

Here is the output from PTG:
```
2019-09-16 08:45:38,901 Result: {"printingArea": {"maxZ": 50.00004577636719, "maxX": 110.75, "maxY": 140.75, "minX": 60.0, "minY": 10.0, "minZ": 0.0}, "dimensions": {"width": 50.75, "depth": 130.75, "height": 50.00004577636719}, "firstFilament": 0.00061323437595528, "progress": [[0, 389.0351849893795], [0.00044186223161733, 383.9147957376699], [0.00061323437595528, 378.5106095914033], [0.14702896281872904, 318.4677086969341], [0.30870514913081903, 258.43112231628993], [0.47039893582530046, 198.39375233247654], [0.6329690362956938, 138.38551009983487], [0.7946711600134133, 78.341790180641], [0.9563899577776089, 18.32767698734665], [0.9995785171585202, 2.620048777579939], [0.9999796206098626, 0.0], [1, 0]], "estimatedPrintTime": 389.0351849893795, "lastFilament": 0.9995785171585202, "filament": {"tool0": {"volume": 1.5571815453419153, "length": 647.4008381023529}}}

2019-09-16 08:45:38,903 Merged result: {u'printingArea': {u'maxZ': 50.00004577636719, u'maxX': 110.75, u'maxY': 140.75, u'minX': 60.0, u'minY': 10.0, u'minZ': 0.0}, u'dimensions': {u'width': 50.75, u'depth': 130.75, u'height': 50.00004577636719}, 'analysisPending': True, u'firstFilament': 0.00061323437595528, u'progress': [[0, 389.0351849893795], [0.00044186223161733, 383.9147957376699], [0.00061323437595528, 378.5106095914033], [0.14702896281872904, 318.4677086969341], [0.30870514913081903, 258.43112231628993], [0.47039893582530046, 198.39375233247654], [0.6329690362956938, 138.38551009983487], [0.7946711600134133, 78.341790180641], [0.9563899577776089, 18.32767698734665], [0.9995785171585202, 2.620048777579939], [0.9999796206098626, 0.0], [1, 0]], u'estimatedPrintTime': 389.0351849893795, u'lastFilament': 0.9995785171585202, u'filament': {u'tool0': {u'volume': 1.5571815453419153, u'length': 647.4008381023529}}}
2019-09-16 08:45:39,120 Disabled: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/single_wall_cylinder_layer.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining
2019-09-16 08:45:39,123 Compensated result: {'analysisFirstFilamentPrintTime': 10.524575397976207, u'printingArea': {u'maxZ': 50.00004577636719, u'maxX': 110.75, u'maxY': 140.75, u'minX': 60.0, u'minY': 10.0, u'minZ': 0.0}, u'dimensions': {u'width': 50.75, u'depth': 130.75, u'height': 50.00004577636719}, 'analysisPending': False, 'analysisPrintTime': 389.0351849893795, 'analysisLastFilamentPrintTime': 386.41513621179956, u'firstFilament': 0.00061323437595528, u'progress': [[0, 389.0351849893795], [0.00044186223161733, 383.9147957376699], [0.00061323437595528, 378.5106095914033], [0.14702896281872904, 318.4677086969341], [0.30870514913081903, 258.43112231628993], [0.47039893582530046, 198.39375233247654], [0.6329690362956938, 138.38551009983487], [0.7946711600134133, 78.341790180641], [0.9563899577776089, 18.32767698734665], [0.9995785171585202, 2.620048777579939], [0.9999796206098626, 0.0], [1, 0]], u'estimatedPrintTime': 389.0351849893795, u'lastFilament': 0.9995785171585202, u'filament': {u'tool0': {u'volume': 1.5571815453419153, u'length': 647.4008381023529}}}
```

I don't think that the issue is with heating/cooling times since the predicted extruding time is 375.9 and the actual extruding time is 689.1 in the "compensation value" table.

Note that linear advance parameter K is set to 0 in the firmware config, but set to the actual value in the gcode. Does PTG send it (M900) from the gcode file to marlin-calc?

eyal0, 2019-09-16:
Okay, I found that I am able to set the linear advance setting and it has an effect.  So that's better.

No, PTG does not send M900 from the gcode to marlin-calc.  The codes that are supported are [here](https://github.com/eyal0/OctoPrint-PrintTimeGenius/blob/master/octoprint_PrintTimeGenius/printer_config.py#L131).  No M900 but that would be easy to add.

`marlin-calc` ignores `M900` right now.  I can fix that pretty easily.

Just to make sure that I got it right, can you tell me what `M900` value you have?  Using the gcode that you provided above along with different values of M900, here's what I have seen.  Let me know if it looks right.  If it does, then I'll get to work on it.

| M900 value | Estimated print time in seconds |
|------------|----------------------|
| 0          | 389                  |
| 0.25       | 400                  |
| 0.5        | 431                  |
| 0.75       | 496                  |
| 1          | 551                  |
| 1.25       | 598                  |
| 1.5        | 640                  |
| 1.75       | 678                  |
| 2          | 714                  |

Also, you're pretty sure that M900 set to 0 is the same as linear advance being disabled?

gpouger, 2019-09-16:
I'm 100% sure that "M900 K0" disables linear advance, it actually says so in Configuration_adv.h

The K factor that I used for this print is 0.6 (it's in the gcode file, line 10).
According to your table it should produce an estimate of roughly 450s, which is still quite far from the reality (689.1 extruding time).

I checked the values passed to marlin-calc, all looks good so I don't know where the difference may be...
Maybe it has to do with the fact that I'm still on Marlin 1.1.9?

gpouger, 2019-09-16:
I just tried printing the same gcode without linear advance (K=0) and I got 603.9 seconds of extruding time in PTG's compensation table. There is definitely something else going on...

gpouger, 2019-09-17:
I think I understand what's going on: I tried Marlin 2.0.0 with and without Junction Deviation, I get print times similar to PTG's estimations when Junction Deviation is enabled.
But on the other hand the print quality is really crappy, I couldn't find a set of parameters that give me correct print results with Junction Deviation enabled.

I don't think many people use it by default, so maybe it shouldn't be enabled in marlin-calc. What do you think?

eyal0, 2019-09-17:
Okay, great!  So we've sorted out some of that at least!  You're seeing that with Marlin 2.0.0 on your printer, with junction deviation, you're getting times that are much closer to that of PTG.  Great.

But you don't like junction deviation.  I'm not one to judge!

I think that I put the junction deviation in there as optional, so it's supposed to be enabled only if your settings indicate that it's enabled!  It's supposed to work like this:

If you have an `M205` command that sets the `J` setting, then junction deviation is turned on.  Otherwise, if you have any of the XYZE settings, jdev is off.  And it should work just like without jdev.

So I expect that both people with and without jdev, it's accurate because both are supported.  What print times did you see with and without jdev and what estimates did you see with and without?

Even though the estimation is based on running actual marlin software, for some reason it isn't perfect and I don't know why.  I'd expect it to be exact but maybe there are some factors making the actual different from the expected.  That's why there is the compensation.

Putting marlin-calc on the old Marlin version would be difficult.  Would you be able to move to the new Marlin 2.0 version?

gpouger, 2019-09-17:
I'm still on 2.0, without junction deviation it works as fine as 1.9 for me.
I had a quick look at the code, and both in config.h and in M200-M205.cpp it seems to me that junction deviation and jerk are exclusive at compile time. It would be nice if it worked as you describe though...

When I enabled junction deviation I got a 1.5 seconds difference between actual and prediction over a 100s print, probably because of linear advance.
Without junction deviation I get 218.9 actual extrusion vs 159.8 estimated.

I don't have any M205 codes in my GCODE file, only what PTG sends to marlin-calc at the beginning. I'll make some more tests toorrow with some M205 in the files.

eyal0, 2019-09-18:
> I'm still on 2.0, without junction deviation it works as fine as 1.9 for me.
> I had a quick look at the code, and both in config.h and in M200-M205.cpp it seems to me that junction deviation and jerk are exclusive at compile time. It would be nice if it worked as you describe though...

Official Marlin only includes one or the other because they can't be used at the same time.  The reason that they are compiled separately is to keep the image size small so that it fits on limited architectures.  `marlin-calc` has both built-in and selects jdev if it sees a jdev setting in use.  I'll do the same technique for linear advance.

> When I enabled junction deviation I got a 1.5 seconds difference between actual and prediction over a 100s print, probably because of linear advance.
> Without junction deviation I get 218.9 actual extrusion vs 159.8 estimated.

Weird.  I wonder if that's just a coincidence.  So that I've got it straight: marlin-calc estiamte with jdev more closely matches your actual print time without jdev then marlin-calc estimate without jdev.  Right?

I wonder if it's just a fluke.  Could we try, for example, print once with jdev, once without, then print once with jdev, once without, and see if jdev is consistently a longer print time.

Anyway, adding linear advance into marlin-calc is not hard to do and it's the right thing to do, so how about I do that first and then we try to sort out why marlin-calc isn't exactly right.

I'd like to fill out a table like this:

| jdev setting | in advance | estimate | actual |
|------------|----------------------|----|---|
| 0 | off                | | |
| 0.2 | off                | | |
| 0 | ON                | | |
| 0.2 | ON                | | |
| etc | | |

And even if the values aren't spot-on, we expect that the *relative* values are correct.  Like, if `A` is slower than `B` in estimate then it should also be slower in actual, etc.  And maybe it would show that one setting is especially causing trouble.  Stuff like that.

gpouger, 2019-09-18:
Seems that I wasn't quite clear in my explanations, sorry about that.

In the end, when I enable jdev in my firmware I get print times very close to the estimations.
When I don't have jdev enabled in my firmware my effective print times are longer, but PTG still computes the same estimates as if jdev was enabled.

Also I checked, if I add an M205 X7 Y7 in my file the estimate changes.

So I think that the "real" problem is that even though PTG gives the initial "M205 B20000.00 E5.00 S0.00 T0.00 X7.00 Y7.00 Z0.20" to marlin-calc, if the parsed file doesn't contain M205 then the estimation is still computed with jdev.

I hope that it makes sense this time.

eyal0, 2019-09-18:
Oh wow, that is a big problem!  The inputs that you put on the command line should be processed exactly the same as the ones that provided in the file.  Essentially, the command-line options are processed as if they are pasted to the very start of the file and then processing proceeds.  The timing results should be the same, the only difference is that the file positions would be different.  So, like, if the print is 100s long at line number 1234, that's 1234 lines not counting the lines in the command-line.  It's important for PTG to be able to report the time remaining at each point in the file.

I'll try this out and see what's gone wrong!

gpouger, 2019-09-18:
Also please add Linear Advance ;)

eyal0, 2019-09-24:
Want to test it before I make a release?  You can install this directly into OctoPrint from the usual plugin dialog:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/linear_advance.zip

It does two things:

1. Adds supports for linear advance.  You might need to run `M205 ` just once so that PTG can pick up your M900 setting.  From there on out, it'll remember it and update it as needed.
2. Fixes a small but horrific bug that I had in marlin-calc.  I was totally ignoring acceleration in the math so there may be all sorts of errors in estimates.  With this fixed, I hope that estimates will look closer to the actual value.  Unfortunately, I don't have a database of known, actual print times so it's impossible for me to know if my new results are better or worse!  If you have some way to figure out if the new marlin-calc is doing better than the old one, I'm all ears.

eyal0, 2019-09-25:
I just updated it again.  Same URL, now the calculation should be even better.

gpouger, 2019-09-25:
I'll try to run a few small test prints later today. Thanks for the quick improvement!

eyal0, 2019-09-25:
Thanks!  Please check, for example, that the estimated print time is affected by the M900 option that you have, whether in the gcode or saved in the config.  Also, the M900 should be taken if you type it into the command line or it's in your saved config.

Once it looks good to you, I'll make it an official release.

gpouger, 2019-09-25:
First round of tests, changing the Linear Advance parameter definitely changes the estimate, but I still have huge differences between the estimate and real printing time:
<table>
  <thead><td>LinAdv in file</td><td>with retracts</td><td>Estimate (extruding)</td><td>Real (extruding)</td></thead>
  <tr><td>0.6</td><td>yes</td><td>440.1</td><td>704.2</td></tr>
  <tr><td>0</td><td>yes</td><td>369.7</td><td>606.4</td></tr>
  <tr><td>0.6</td><td>no</td><td>402.5</td><td>627.1</td></tr>
</table>

As you can see there is a 30% to 40% difference between the estimate and real life print time. I tried removing the G10/G11 retracts from the file just in case but it doesn't help.

Attached is my test file (I manually update it for changing/removing the M900 and the G10/G11 lines).
[test_LinAdv_0.6.gcode.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3654913/test_LinAdv_0.6.gcode.txt)

And here are the initial M-codes that PTG sends to marlin-calc:
M200 D1.75
M200 D0
M92 X80.00 Y80.00 Z1600.00 E415.00
M203 X300.00 Y300.00 Z7.00 E70.00
M201 X3000.00 Y3000.00 Z100.00 E10000.00
M204 P1000.00 R2000.00 T1500.00
M205 B20000.00 E5.00 S0.00 T0.00 X7.00 Y7.00 Z0.20
M207 F2700.00 S3.00 W13.00 Z0.00
M208 F480.00 S0.00 W0.00
M900 K0.00

eyal0, 2019-09-26:
It's odd.  I don't understand it.  The code that I'm using to do my estimates is a hardly modified Marlin.  It does the same thing that Marlin does expect that, at the points that Marlin waits for clock cycles to pass, it just adds up those clock cycles.  So the times should be the almost identical except for perhaps places where the timer is off by a few cycles.

Hmm.  One of the major things that marlin-calc does that I had to write is the parsing of all those m-codes.  I noticed that when I modify the `M205 B` value, it doesn't have any effect on the calculation.  Yet, it should from what I've read.

I'll add support in for that and see if it makes a difference.  If you want, try to mess with other settings that you think might be timing related and see if they are having an effect and in the correct direction.  Like, things that should make printing slower ought to make the estimate longer.  Stuff like that.

eyal0, 2019-09-26:
I spoke too soon about `M205 B`.  With a really large value, it is having an effect.

Want to paste here your M503 output and we can see if you have any settings that ought to have an effect yet aren't?

eyal0, 2019-09-26:
Last thing: At least the compensation would be easy.  For those 3 prints, they are all off by a factor of around 1.6.  PTG would figure out that compensation and so you'd still get decent answers.  However, maybe compensation would be different for different models.

Though we're still seeing wrong answers, I think it's still an improvement because linear advance is supported and I did fix a pretty bad bug, so I think that I'm going to push the new version while we keep looking at the timing.

I wonder if the timing issue has to do with the shape of the object or the direction...?  Here's some gcode that just moves the print head back and forth.  Pretty simple.  I wonder if it, too, would have a factor of 1.6 difference?

```
G92 X0 Y0 Z0
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
G1 X0 Y0 Z0 F600
G1 X100 Y0 Z0 F600
```

PR #157 (2019-09-26): Add M900 support
This fixes #153 

`M900 K<param>` is saved in the config and the new version of marlin-calc supports it.
closed 2019-09-26 by eyal0