eyal0/OctoPrint-PrintTimeGenius #154: Specifying options on the command-line should have the same effect as putting them in the file

eyal0, 2019-09-18:
Seems that I wasn't quite clear in my explanations, sorry about that.

In the end, when I enable jdev in my firmware I get print times very close to the estimations.
When I don't have jdev enabled in my firmware my effective print times are longer, but PTG still computes the same estimates as if jdev was enabled.

Also I checked, if I add an M205 X7 Y7 in my file the estimate changes.

So I think that the "real" problem is that even though PTG gives the initial "M205 B20000.00 E5.00 S0.00 T0.00 X7.00 Y7.00 Z0.20" to marlin-calc, if the parsed file doesn't contain M205 then the estimation is still computed with jdev.

I hope that it makes sense this time.

_Originally posted by @gpouger in https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/153#issuecomment-532716248_

eyal0, 2019-09-20:
@gpouger I tried reproducing this problem but I didn't see it.  Using the file from [here](https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/153#issuecomment-531775192), I ran this:

```
"./octoprint_PrintTimeGenius/analyzers/marlin-calc.x86_64" ~/Downloads/single.wall.cylinder.layer.gcode   "M200 D1.75
M200 D0
M92 X80.00 Y80.00 Z1600.00 E415.00
M203 X300.00 Y300.00 Z7.00 E70.00
M201 X3000 Y3000 Z100 E10000
M204 P1000.00 R2000.00 T2500.00
M205 B20000 E5.00 S0.00 T0.00 X7.00 Y7.00 Z0.20
M207 F3600.00 S3.00 W13.00 Z0.00
M208 F3600.00 S0.00 W0.00" | tail
```

And then again with this added as the first line of the file:

```
M205 X7 Y7
```

There is no effect on the total time output.  Did I do the procedure right?  Can you try it again and figure out how I can reproduce this issue?

gpouger, 2019-09-25:
Hi, don't know what I did wrong last time but I played with it again and it behaved as expected (no difference with or without M205 in the file, jdev disabled by default).

Maybe I messed up the initial M-codes with too many tests and different Marlin versions...

Sorry for the false alert!

eyal0, 2019-09-25:
No problem.

closed 2019-09-25 by eyal0