eyal0/OctoPrint-PrintTimeGenius #158: Estimates are off by a factor of 1.6

eyal0, 2019-09-26:
First round of tests, changing the Linear Advance parameter definitely changes the estimate, but I still have huge differences between the estimate and real printing time:
<table>
  <thead><td>LinAdv in file</td><td>with retracts</td><td>Estimate (extruding)</td><td>Real (extruding)</td></thead>
  <tr><td>0.6</td><td>yes</td><td>440.1</td><td>704.2</td></tr>
  <tr><td>0</td><td>yes</td><td>369.7</td><td>606.4</td></tr>
  <tr><td>0.6</td><td>no</td><td>402.5</td><td>627.1</td></tr>
</table>

As you can see there is a 30% to 40% difference between the estimate and real life print time. I tried removing the G10/G11 retracts from the file just in case but it doesn't help.

Attached is my test file (I manually update it for changing/removing the M900 and the G10/G11 lines).
[test_LinAdv_0.6.gcode.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3654913/test_LinAdv_0.6.gcode.txt)

And here are the initial M-codes that PTG sends to marlin-calc:
M200 D1.75
M200 D0
M92 X80.00 Y80.00 Z1600.00 E415.00
M203 X300.00 Y300.00 Z7.00 E70.00
M201 X3000.00 Y3000.00 Z100.00 E10000.00
M204 P1000.00 R2000.00 T1500.00
M205 B20000.00 E5.00 S0.00 T0.00 X7.00 Y7.00 Z0.20
M207 F2700.00 S3.00 W13.00 Z0.00
M208 F480.00 S0.00 W0.00
M900 K0.00

_Originally posted by @gpouger in https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/153#issuecomment-535234092_

gpouger, 2019-09-26:
Here is the output of M503:

Send: M503
Recv: echo:  G21    ; Units in mm (mm)
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo: M92 X80.00 Y80.00 Z1600.00 E415.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X300.00 Y300.00 Z7.00 E70.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P1000.00 R2000.00 T1500.00
Recv: echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 B20000.00 S0.00 T0.00 X7.00 Y7.00 Z0.20 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Material heatup parameters:
Recv: echo:  M145 S0 H205 B70 F0
Recv: echo:  M145 S1 H240 B110 F0
Recv: echo:PID settings:
Recv: echo:  M301 P30.55 I2.65 D88.18
Recv: echo:  M304 P140.00 I30.00 D200.00
Recv: echo:Retract: S<length> F<units/m> Z<lift>
Recv: echo:  M207 S3.00 W13.00 F2700.00 Z0.00
Recv: echo:Recover: S<length> F<units/m>
Recv: echo:  M208 S0.00 W0.00 F480.00
Recv: echo:Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:Filament load/unload lengths:
Recv: echo:  M603 L500.00 U600.00
Recv: ok

gpouger, 2019-09-26:
Just uploaded your GCODE that moves the printhead around without extruding and PTG failed to estimate it, here is the log.
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3656501/plugin_PrintTimeGenius_engine.log)

And the actual file:
[test.gcode.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3656508/test.gcode.txt)

eyal0, 2019-09-26:
PTG expects extrusion. When there is none, the analysis fails.

My idea was to send gcode commands that have a known time. For example, 100mm at 600mm/minute should take exactly 10 seconds. We can set the printer to do that, watch it with a stopwatch, and see if it really takes 10 seconds. Similarly for drawing a circle, etc.

Maybe through tests like that, we could figure out why your prints are slow by a factor of 1.6. Alternating `G1 X0 F600` and `G1 X100 F600` should have your printer switch direction every 10 seconds. Is that right?

Then the next step would be to try it in a circle, I guess.

gpouger, 2019-09-30:
I'm starting to think that part of the problem is that my test file is sort of a "worst case scenario": it's a single wall cylinder, and I'm wondering if printing it from USB is problematic (lots of short moves, maybe the USB link or my 8-bit board is not fast enough)...
I'll make more tests with simpler moves.

gpouger, 2019-10-01:
I made some tests and played with my Marlin config, I got to a point where the same test print (the single wall cylinder) got printed with a 1.2 difference factor instead of 1.6
I think this proves that my printer is constantly stuttering when printing from USB...

On the other hand, I have a much simpler test (single wall square, 4cm wide and 5mm high) that gets printed in 120.4 seconds for a 119.9 estimation (again, only looking at the extrusion times).
Looks like when my printer is actually running at the speed that it should, the estimation of the last PTG is very close - at least on this very simple model.

I'll try to design a model with long enough edges to avoid the stuttering, and different angles to actually test the computation of Linear Advance.

eyal0, 2019-10-01:
That's some good investigating, thanks!

Another thing to try if you haven't yet, you could lower the acceleration by *a lot*, to the point where it's affecting print speed on that single-wall square.  That might be interesting.  `marlin-calc` is adding up the area under a trapezoid that is made from starting at an initial speed, accelerating to cruising speed, cruising for a while, and then decelerating back to the initial speed.  As the `M205` acceleration speed is lowered, more and more time is spent changing speed.  I think that I got the calculation of those speed changes correct but it could use testing.

gpouger, 2019-10-01:
I tried your suggestion, printed my single wall square with small accelerations (100 instead of 1500), and the estimate was accurate enough for me (optimistic by a factor of 1.06).

I also ran a real model (3DBenchy) and ended up with a factor of 1.05, so also quite close (5 minutes difference over a 1 hour print).

My conclusion is that the computations in marlin-calc are quite good, but you can't account for the time spent by the mainboard doing computations or for slowing down due to empty buffer.

Note that I tested with Linear Advance but none of the fancy new options (junction deviation, S-Curve acceleration, adaptive step smoothing etc).
I'll give it an other go with these options when  I get a 32-bit board (on the way from China).

eyal0, 2019-11-13:
Waiting on you to provide more info.  If you see a specific timing measurement error, let me know.

closed 2019-11-13 by eyal0

gpouger, 2019-11-25:
Since I switched to a 32-bit board the estimations are pretty much spot-on, so it seems that the problem was just my 8-bit board struggling and taking more time to do its job than it should.
No more complaints from me, good job!