eyal0/OctoPrint-PrintTimeGenius #159: Plugin never completes analysis

ProfEngr, 2019-09-27:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.3.11

PrintTimeGenius Version (if you know): 2.0

What did you try: disable then enable plugin. uninstall/reinstall.

What happenned: the gold star in the file list just keeps blinking even after leaving it overnight

What did you expect to happen: there was a lot more log output in the previous version

[plugin_PrintTimeGenius_engine.log.2019-09-25-1.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3663234/plugin_PrintTimeGenius_engine.log.2019-09-25-1.txt)
[plugin_PrintTimeGenius_engine.log-1.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3663235/plugin_PrintTimeGenius_engine.log-1.txt)

vadsh, 2019-09-27:
I have the same issue

556duckvader, 2019-09-28:
Same here and almost makes Octoprint not work.

nicodemus, 2019-09-28:
Same issue here. In my case, marlin-calc.armv7l sits at 100% cpu usage and never finishes analyzing the file.

hannoHI, 2019-09-28:
Same problem.

alsand61, 2019-09-28:
Same problem here, after updating plugin to 2.0

ProfEngr, 2019-09-28:
I've also noticed an issue with auto-reconnect not being automatic. I get errors about exclusive access to the USB port. Not sure if this is related, but it's also just become an issue.

Gacky2k, 2019-09-28:
Yup..same issue here...won't finish analyzing.   :(

eyal0, 2019-09-28:
Shit. It worked fine in my testing. I'm cross-compiling from a new computer so maybe I got the libraries wrong?

Send a gcode file on which marlin-calc never completes please and I'll dig into it. Unfortunately I don't have an RPi on which to test so I can only test the x86 version for now.

ProfEngr, 2019-09-29:
This has linear advance, junction deviation, and S curve enabled if that matters.
[Guitar_Pick.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3666157/Guitar_Pick.zip)

Gacky2k, 2019-09-29:
Does this help?
[octoprint.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3666170/octoprint.log)

eyal0, 2019-09-29:
I'm seeing it on the Guitar_Pick.gcode file now, too.  I'll see what's up.

closed 2019-09-29 by eyal0

eyal0, 2019-09-29:
Having the gcode file made it easy to fix.  Thanks.  Please try v2.0.1 and let me know how it performs.  Estimates ought to be better.  Reopen this if you continue to see problems, thanks!

Get v2.0.1 the usual way or use this [link](https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/2.0.1.zip).

Gacky2k, 2019-09-29:
Didn't freeze this time!  With v2.0.1 that is.  But....always a but  :)   I loaded a print to see what the default analyzer would show (Genius was still disabled at this time) and it estimated 8h and some change.  PrusaSlicer estimates it at 10h and change.  Upgraded to v2.0.1, re-enabled it and reuploaded the same file,and Genius estimates it at over 35h.  

We'll see if it's just a first print fluke or what happens to that estimate as it goes.  I'll post more results as it prints.

eyal0, 2019-09-29:
That might be because I fixed something in marlin-calc so your old compensation values from before v2.0 aren't accurate anymore for you.  You can reset the compensation by deleting all the rows in the Avanced PTG options.  Or just keep printing and after 5 prints the new old compensation values will have aged out.  You can see the pre-compensation timing in the log.  It might be closed to the 8-10hours estimate.

Or if you send the gcode file, I can run it for you and give you the answer, same as what the log will say.

eyal0, 2019-09-29:
@Gacky2k also follow #158 where I'm trying to figure out better accuracy. If you want to participate in helping, let me know!

Gacky2k, 2019-09-29:
I deleted the rows, but I'm in the middle of a ~10h print (PrusaSlicer estimate), so I wont be able to test if that helped for a bit yet.  The estimate is updating on the fly though.  Now, after 5h into what it thought originally was going to be 35h, it is now saying 14h remaining.

PrusaSlicer on my i3 MKS has been pretty damn accurate, usually within a few minutes, but never over 30 min's off, and that's on really long prints.  I really utilize this plugin mostly for estimating/planning prints on my Ender 3 Pro.  I use Cura for slicing on it as I have tweaked profiles on that that work really well and I'm too lazy to create one from those profiles inside PrusaSlicer :)  Cura used to be HORRIBLY awful at estimates, by as much as 50%.  Most recent version is much better.

I'd be happy to help in any testing.  I can clear out those rows on my OctoPrint on my Ender 3, and update to v2.0.1 and do whatever other testing you would like me to.

eyal0, 2019-09-30:
@Gacky2k So if you really like PruseSlicer's estimates, you can have those override marlin-calc's.  You can just enable the M72 Slic3r PE estimates, it's an option in the settings of PTG.

I'm curious to see if the latest marlin-calc can be as good as the slic3r PE estimates.  I'm really clock-for-clock accurate on the latest version, I think.  If it isn't really close, I'd like to try to debug.  I had a thought to create some known shapes, like a cylinder, and I would calculate by hand the time that I expect it take.  Then run it on a printer and see what happens.

Anyway, let me know how the new version goes.  I really did have a horrible error per v2.0 and I hope that this new version will be much better.  I hope that I can get the compensation down to 1-to-1, that is, no adjustment needed, the estimates are just perfect.  The best thing is if you can provide, like, the actual time spent extruding (not counting heatup and cooldown) and the estimates that marlin-calc gives before compensation.  Those are what I'd like to be 1-to-1.

Gacky2k, 2019-09-30:
I saw that option, to use M72 Slic3r PE estimates, but wasn't sure if it was compatible, now that Prusa has rebranded it to just Prusa Slicer and have made such HUGE changes with the last couple of releases.  At any rate, Prusa Slicer, by default, will append extra information like layer height and print time, to the end of the file name, so I don't need to change it, this way I can compare the two estimates.

That being said, I have those previously mentioned 2 printers to test on if you have anything you want me to test. (Prusa i3 MK3S, and Ender 3 Pro).  I'd be more than happy to help test.

PR #160 (2019-09-29): Replace other calls to idle with idle2
There remained a call to idle that should have been a call to idle2 in marlin-calc.  It's fixed [here](https://github.com/eyal0/Marlin/commit/58ceb81832661b31a9fbd5241bde5d2c81c876ec).

This fixes #159