eyal0/OctoPrint-PrintTimeGenius #165: PTG fails on all uploaded GCODE

SVLoneStar, 2019-11-08:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.3.12

PrintTimeGenius Version (if you know): 2.0.2

What did you try: Remove some large uploaded GCODE files from Octoprint's Upload folder, manually start analysis in 'Settings', search Google for error, read previous posts on this issue.

What happenned: After manually starting the analysis, marlin-calc.arm gets invoked several times, a log was created and contains the following entries for every single GCODE file uploaded to OctoPrint:
2019-11-08_ 11:31:41,517 Failed to run '"/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/V29D_Fixed.gcode" "M220 S100
M221 S0
M200 D1.75
M200 D0
M92 X80.00 Y80.00 Z400.00 E105.00
M203 X200.00 Y200.00 Z20.00 E50.00
M201 X800.00 Y700.00 Z100.00 E5000.00
M204 P700.00 R800.00 T8000.00
M205 B20000.00 E5.00 S0.00 T0.00 X8.00 Y8.00 Z0.30"'
Traceback (most recent call last):
  **File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 318, in _do_analysis
    raise Exception(results_err)**
Exception: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/V29D_Fixed.gcode" "M220 S100
M221 S0
M200 D1.75
M200 D0
M92 X80.00 Y80.00 Z400.00 E105.00
M203 X200.00 Y200.00 Z20.00 E50.00
M201 X800.00 Y700.00 Z100.00 E5000.00
M204 P700.00 R800.00 T8000.00
M205 B20000.00 E5.00 S0.00 T0.00 X8.00 Y8.00 Z0.30"
Read up all remaining commands
Processed 106744 Gcodes and 132 Mcodes. 96141 blocks
Total time: 7323.804448
**Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 90, in <module>
    main()
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py", line 63, in main
    result.update(json.loads(line))
  File "/usr/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=w(s, 0).end())
  File "/usr/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded**

Of course the exact values of the Mxxx differ, but the bold formatted parts are present for all the 7 objects which are uploaded to OctoPrint. If that makes any difference, most of them are sorted into folders.
Analysis used to work, but stopped some time ago...sorry, no better info when that was and what I did that might have caused the failure. Slicing is (and was) done with Simplify3D.

What did you expect to happen:
Analysis of the uploaded GCODE files.

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).
As the analysis fails for all uploaded files, I attach only one GCODE file here and an axtract of the log.
[V29D_Fixed.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3824207/V29D_Fixed.zip)
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/3824231/plugin_PrintTimeGenius_engine.log)

SVLoneStar, 2019-11-17:
If there's any more information required to analyze this, please let me know....thanks.

dargul, 2019-11-21:
Delete "M200 D0" from your printer init code and you should be fine

SVLoneStar, 2019-11-21:
Thanks a lot for your reply.
I currently have 
#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75
in my Configuration.h and 
//#define NO_VOLUMETRICS
in my Configuration_adv.h.

I did a
M200 D1.75
M500
and started an analysis of the uploaded G-Codes...just to make sure, I checked that do not include a M200 D0.
Same results, same errors in log.
Further ideas? Thanks!

dargul, 2019-11-21:
Your flow rate seems to be set to 0% so maybe a
M221 S100
will solve the problem

SVLoneStar, 2019-11-21:
Thanks a ton!! Sending
M221 S98
M500
and restarting the analysis via PTG's Settings dialog solved the issue!!

Any ideas why or how this got lost in the printer's EEPROM?
In Simplify3D, the value for 'Extrusion Multiplier' is set to '0,98' in all profiles I used to slice the models, but no M221 appears in the models' G-Code. Bug in S3D? Missing default in Marlin?
Anyway...again thanks!

closed 2019-11-21 by SVLoneStar

eyal0, 2019-11-22:
Ah, I should have noticed this.  Yes, the M221 is filament percentage and with 0, it's going to get very confused during the printing.

M200 **is** supposed to show up more than once in the list.  The M200 command is special that M200 D1.75 tells Marlin to use volumetric extrusion and the filament width is 1.75mm.  The second one turns off the volumetric extrusion but the knowledge of the filament width remains.  That's important for computing the volume of filament used, which is reported by OctoPrint.  If you type M503, you'll see that Marlin printers also report first the M200 D1.75 and then the M200 D0.

I should consider making it easy for users to edit settings.  For now, you can dig into ~/.octoprint/config.yaml and adjust it with an editor.  But beware!, the config.yaml file is sensitive to whitespace so make sure that you don't add/remove any blank spaces on accident.

Soon the printer config will be in a new file called printer_config.yaml.

Glad it all worked out in the end.

SVLoneStar, 2019-11-22:
Thanks for the explanations! Just updated to the newest PTG version with printer_config.yaml.
I 'reverted' my M200 settings by sending an M200 D0 via terminal (and an M500 of course). printer_config.yaml now contains (again) both entries. But I think that in the end, M221 S100 did the trick, not the M200 Dxx.
However - it now works again...thanks a lot for this great plugin!