eyal0/OctoPrint-PrintTimeGenius #169: Running PTG during print - is it ok?

mdaneman, 2019-11-22:
This is not an issue report, but a question.  I mainly upload gcode to Octoprint through either Prusa Slicer and Cure, both of which will tell Octoprint to start printing immediately upon upload.  This often doesn't leave enough time (even including heat-up time) for PTG to fully parse the file.  I've so far kept the "Allow analysis while printing" off, because of the scary warning that it may impact the print.  However, what is people's experience letting PTG run during printing?  Is it really a problem or it is ok to do it?  I'm running on a Pi 3B.

Thanks,
-Mike.

eyal0, 2019-11-22:
It might work fine.  Might not.  Pi 3B is pretty quick so it might be okay.  Your best bet is to just turn it on and see what happens.  If you don't notice a problem then it's fine.

I have thought about adding some mechanism that would pause printing until analysis is done but I didn't look into it too much.

calania, 2019-12-09:
> I have thought about adding some mechanism that would pause printing until analysis is done but I didn't look into it too much.

Would personaly really like a feature like that