eyal0/OctoPrint-PrintTimeGenius #171: Needs Updating to Support OctoPrint 1.4.0 and Python 3

PlasmaSoftUK, 2019-12-16:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.4.0rc3

PrintTimeGenius Version (if you know): latest source

I've managed to get this working by modifying 2 files:

__init__.py --> insert a line to explicitly confirm Python 3 compatibility:
679.   __plugin_pythoncompat__ = ">=2.7,<4"

analyse_progress.py --> modify the following lines adding a 'b' to switch from string comparison to byte comparison:

46.   if line.startswith(b"Progress:"):

48.   (filepos, filament, time) = map(float, line.split(b","))

61.   elif line.startswith(b"Analysis:"):

After doing this the plugin installs and appears to work! :)

Thanks

Plasma

eyal0, 2019-12-16:
Sounds great. Want to write a PR?

PlasmaSoftUK, 2019-12-16:
Happy to try that, I guess it's about time I learnt how to make a PR! :D

PlasmaSoftUK, 2019-12-16:
I think I have successfully created a PR!

eyal0, 2019-12-16:
Good job!

closed 2020-01-10 by eyal0