eyal0/OctoPrint-PrintTimeGenius #175: After upgrading to 2.2, print time estimates are half of actual

brettvitaz, 2020-01-08:
OctoPrint Version:
1.3.12

PrintTimeGenius Version (if you know):
2.2

What did you try:
Printing gcode generated from Cura, waited for estimate prior to printing.

What happened:
Print time estimate was 4 hours, actual print time was 8 hours. Cura's estimated time was accurate.

What did you expect to happen:
Estimate to be 8 hours

Note: 
Starting another print now with similar issue of Cura estimates 1 hour, PTG estimates 30 minutes.

eyal0, 2020-01-08:
Please include the PTG log file.  It will show what command marlin-calc ran.

Also, there was recently a big change to the calculation in marlin-calc.  In short, it used to be broken and I fixed it, so all the old compensation values are wrong.  You can remove all the old compensation in the settings, under advanced at the bottom.  Then reanalyze files and it'll work.

If you send me the log of the analysis, I can tell you if that is what happened.

brettvitaz, 2020-01-08:
Here is the PTG log. 
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/4037557/plugin_PrintTimeGenius_engine.log)

I have removed the compensation values and the next calculation was within the expected range. Thank you!

eyal0, 2020-01-10:
Looks like there was some wonky negative number in there.  I don't know how it got in there!  It was causing a lot of lines with `Compensated result` to be around 0.5, which would be the doubling that you saw.

Anyway, I hope that it was just a fluke.  Best of luck!

closed 2020-01-10 by eyal0