eyal0/OctoPrint-PrintTimeGenius #176: Plugin does not install under Python 3 (Raspbian)

Lingomat, 2020-01-11:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.4rc3

PrintTimeGenius Version (if you know): Whatever it is grabbed from the plug-in repository.

What did you try: Installing on 1.4rc3 with Python 3 Venv

What happenned: Fails to install

What did you expect to happen: ... plugin says it's Python 3 compatible, but it's not.

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

It looks like your plug-in is trying to import the logging package on Python 3. It's built-in so this results in the following error on a the Rasbian standard Python 3.7 (in a VEnv). An extract from pluginmanager_console.log:

2020-01-11 17:21:55,501 > from lib2to3.refactor import RefactoringTool, get_fixers_from_package
2020-01-11 17:21:55,501 > File "/usr/lib/python3.7/lib2to3/refactor.py", line 18, in <module>
2020-01-11 17:21:55,501 > import logging
2020-01-11 17:21:55,502 > File "/tmp/pip-install-xp76hip7/logging/logging/__init__.py", line 618
2020-01-11 17:21:55,502 > raise NotImplementedError, 'emit must be implemented '\
2020-01-11 17:21:55,502 > ^
2020-01-11 17:21:55,502 > SyntaxError: invalid syntax

eyal0, 2020-01-11:
This might be. I never tested it. Do you know how to fix it? Or how I can test it so that I can fix it?

Lingomat, 2020-01-12:
I think it would have to be a conditional import based on python version. So just a rough guess, something like:

```
import sys
if sys.version_info[0] < 3:
    import logging
```

That's the main code but the problem is that logging gets marked as a dep. I have no idea how this is done in Python but a quick Google turned up this SO answer that seems to show how to specify Python versions: https://stackoverflow.com/questions/21082091/install-requires-based-on-python-version

eyal0, 2020-01-12:
When I search through the source code for OctoPrint itself, I see many "import logging" in there.  Does OctoPrint support python3?  How is that working?

WolfgangAxel, 2020-01-24:
Python 3 has the `logging` module in the standard library, meaning that you can just `import logging` without needing to manually install a `logging` package. This issue is this: this plugin lists `logging` as a requirement passed through to `pip`. When `pip` tries to install `logging` overtop of `stdlib`'s `logging`, chaos ensues that I don't understand the particulars of. The fix here lies in `setup.py`. Here's a potential, untested suggestion:

    plugin_requires = ["OctoPrint>=1.3.9rc1", "psutil", "sarge"]
    if sys.version_info[0] < 3:
        plugin_requires += "logging"

commit 83d2be43 (2020-02-03): Don't require logging in python3

Fixes #176 
Fixes #180
closed 2020-02-03 by eyal0