eyal0/OctoPrint-PrintTimeGenius #178: [Feature request] Show estimated time

janis-ax, 2020-01-19:
I think it would be nice if it display not only the duration also the estimated time when a print is done. 

E.g.: Duration 3h Estimated time: 10:15 AM

eyal0, 2020-01-19:
That would be possible but it ought to be in a different plugin, because it is functionality that doesn't depend on PTG.  So there could be PTG, providing good estimates, and another plugin, which causes the estimate to display as a time.

There might already be plugins for that, though.  I know that there was some plugin that would send ETA to the printer, it was called M73 or M74 or something...

eyal0, 2020-02-04:
Here's one: https://plugins.octoprint.org/plugins/octoprint-eta/

That will put the "ETA" on the screen.  Try that and let me know how it integrates together with PTG.

closed 2020-02-04 by eyal0

janis-ax, 2020-02-04:
> 
> 
> Here's one: https://plugins.octoprint.org/plugins/octoprint-eta/
> 
> That will put the "ETA" on the screen. Try that and let me know how it integrates together with PTG.

Yeah, work's fine !!