eyal0/OctoPrint-PrintTimeGenius #180: SyntaxError on Python 3

Steeven9, 2020-02-03:
OctoPrint Version: 1.4.0rc4

PrintTimeGenius Version (if you know): latest from the GitHub link

What did you try: Install the plugin via the GitHub URL

What happenned: Python 3 gives a SyntaxError

What did you expect to happen: a successful installation

Below is the installation log with the error:
[log.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/4150251/log.txt)

eyal0, 2020-02-03:
Maybe this?

https://stackoverflow.com/questions/32386469/logging-module-not-working-with-python3

What did octoprint do to solve this problem?

eyal0, 2020-02-03:
Fixed by #178 ?

closed 2020-02-03 by eyal0

Steeven9, 2020-02-03:
Can confirm, successful install.
Thank you very much!

eyal0, 2020-02-04:
On version 2.2.1? Hooray!

Steeven9, 2020-02-04:
Yup, version 2.2.1 running on OctoPrint 1.4.0rc4, win7 32bit, python 3.8.1. 
Good job!

commit 83d2be43 (2020-02-03): Don't require logging in python3

Fixes #176 
Fixes #180