eyal0/OctoPrint-PrintTimeGenius #183: Request: option to disable gold star

timur-tabi, 2020-03-08:
Please add an (advanced) option to disable the gold star.  I can see now that PrintTimeGenius is working.  I don't need a gold star next to every file name any more.

eyal0, 2020-07-13:
Can you test this please?  You can install this in OctoPrint using the normal plugin installation method with this link:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/showStars.zip

The option is in the settings dialog.

timur-tabi, 2020-07-13:
I only have one file in my queue, but this version does appear to work.  The gold star is gone.  Thanks.

jneilliii, 2020-07-15:
I can confirm @eyal0 this version does allow for removing the star (thanks so much) and did resolve the issue with 1.4.1RC2.

PR #204 (2020-07-15): Add showStars option.
This allows users to disable the gold stars in the file list.

This fixes #202
This fixes #183
commit 3337c570 (2020-07-15): Add showStars option.

This allows users to disable the gold stars in the file list.

This fixes #202
This fixes #183
closed 2020-07-15 by eyal0