eyal0/OctoPrint-PrintTimeGenius #185: [Feature Request] Pause timer during tool change

ewortzman, 2020-03-29:
OctoPrint Version: 1.4.0

What did you try: Ran a print during which my filament runout detector paused the print until I could change the filament

What happenned: the timer continued timing the print, causing the compensation value to grossly overestimate for the next few prints

What did you expect to happen: the timer would pause when the printer is waiting for manual user input, such as a manual filament change

I had a print that PTG estimated at about 6 hours, and so I started it and went to bed.  In the middle of the night, my filament ran out and my runout detector stopped the print and waited for a filament change.  However, I didn't wake up until several hours later, and as a result, the total elapsed time of the print was around 14 hours.  PTG then used this 14 hour time in its compensation table, resulting in the next several prints grossly overestimating. In one instance, I had a print that PTG estimated at 9 hours, which actually only took about 100 minutes.

It would be nice if PTG could detect something like an M600 and pause its internal timer until the user executes the resume command, such as pressing the LCD button or executing M108.  That way, it will not overcompensate future prints when there is an unexpected (or expected) pause in the print job, and instead merely estimates actual print time.

eyal0, 2020-04-16:
Yes, you're not the first to request this.  Way back in #92 people saw this.

I'm not quite sure how to fix it yet.  I'll keep thinking about it.  If I had a solution, would you be able to test it?

eyal0, 2020-04-16:
(Because I don't have a printer right now so I am unable to test.)

ewortzman, 2020-04-16:
Absolutely! If you think of something, let me know!