eyal0/OctoPrint-PrintTimeGenius #186: Logs Show Perfect estimations, Octoprint Time Left Never Gets Updated

adawalli, 2020-04-11:
OctoPrint Version: 1.4.0

PrintTimeGenius Version (if you know): 2.2.1

What did you try:
Files appear to be analyzed correctly, appropriate estimations show up in Genius logs

What happenned:
The UI is never updated with Gold Star (or the calculated remaining time). The Gold Star does show up next to files in the upload list

What did you expect to happen:
I want to see estimated times on the UI 

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

[CE3_Surgical_Mask_Strap_Remix_v3.gcode.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/4464955/CE3_Surgical_Mask_Strap_Remix_v3.gcode.txt)
[octoprint.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/4464967/octoprint.log)
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/4464968/plugin_PrintTimeGenius_engine.log)

adawalli, 2020-04-14:
@eyal0 - let me know if you need more detailed information and I will be happy to provide!

eyal0, 2020-04-16:
I don't see anything useful in those logs to help me debug.  I know that if you start printing during the PTG calculation, the printing can cancel the calculation.  Can you try to go into the PTG settings, calculate print time for a file, wait for it to get the gold star, and then print that and see if it gets PTG estimations?

adawalli, 2020-04-16:
I am not getting any cancellations in the logs. I purposely have tailed the logs and waited until PTG finishes (the goldstar stops flashing in the file browser as well and goes solid) and yet still never propagates to the actual estimate window. Anything else I can do to help triage?

eyal0, 2020-04-28:
So you see the gold star flashing in the list of files?  Do you also see gold stars next to files that have been analyzed and finished being analyzed?

While you print, I would hope that there would be a gold star next to where the print estimate is, instead of the usual red yellow or green dot.  You're not getting that one?

adawalli, 2020-04-28:
- I see a Gold Start Flashing when the file is being analyzed
- I see a solid gold star when the file is finished being analyzed
- I **NEVER** see a gold star in the status indicator. I get a green/yellow/red dot

eyal0, 2020-04-28:
Ah, that's no good.  Hmm.  While printing, are you seeing anything in the PTG log?  It's also possible to enable debugs for more verbose logs, maybe that will be useful?

Instructions on how to do that here: https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

After you do it, you ought to see more in the logs while you print.  It can also affect print time and quality so you might want to do it just for testing purposes and then turn it back off.

adawalli, 2020-04-28:
Hi there - I attached the DEBUG logs in my original question above - was information missing in those logs?

Wargameskhaine, 2020-08-14:
You are not running plugin "M73 ETA Override" per chance? I had a similar issue and narrowed it down to this.

adawalli, 2020-08-14:
Hmm. In fact I am.

coryflucas, 2021-05-19:
I'm seeing this error in the logs when printing files estimated with the Slic3r PE M73 time remaining analyzer:
```
2021-05-19 12:42:46,093 Failed to estimate, ignoring.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 145, in estimate
    statisticalTotalPrintTime, statisticalTotalPrintTimeType)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 126, in _genius_estimate
    progress <= self._metadata["analysis"]["lastFilament"]):
KeyError: 'lastFilament'
```
I analyzed files with the both the Slic3r PE and the marlin calc analyzer, and the marlin analyzer adds a "lastFilament" property in the .metadata.json file, but the Slic3r PE one doesn't.

eyal0, 2021-05-19:
@coryflucas is this related to the original issue? If not, you can open a new issue.

Maybe I forgot to put lastFilament into the slic3r analyzer. That could be. I'll take a look...

coryflucas, 2021-05-19:
Sorry, yes I wasn't super clear.  I'm seeing the same behavior where it says files are analyzed, but the analysis is not showing during prints which is when I see that error in the logs. As others have reported I don't see errors during analysis of files.

eyal0, 2021-05-19:
Good to know.

If you disable the slic3r M73 analyzer and use just the marlin-calc one, does the problem go away?

If you can, please send your gcode file, too.

coryflucas, 2021-05-19:
Confirmed that if I upload a file with the slic3r analyzer on, it is analyzed (gets the star next to it in the file list), but during printing the estimate is not used (no star). After deleting the file, enabling the marlin analyzer instead, uploading the same file results in it being analyzed and the estimate being used during printing.  Confirmed the gcode contains M73 commands, and can be found here: https://gist.github.com/coryflucas/40a8bf87b7ba789b8b557349b623aecc.

BTW thanks for the great plugin and all your help looking into this!

eyal0, 2021-07-08:
Sounds like we're okay now.  Thanks.

closed 2021-07-08 by eyal0