eyal0/OctoPrint-PrintTimeGenius #187: I'm not getting any gold stars. I don't understand what is supposed to happen.

spierepf, 2020-04-22:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

I'm not getting any gold stars.

The log contains:
```
2020-04-21 14:51:21,875 Starting PrintTimeGenius
```
and thats all

Please fill this out:

OctoPrint Version: Version 1.4.0

PrintTimeGenius Version (if you know): I can't find it anywhere on my OctoPi, but I just installed it a few hours ago.

What did you try: Following the instructions at: https://www.youtube.com/watch?v=ozCqqlPJ3a0&t=203s

What happenned:

The plugin is installed, but nothing really changes.

What did you expect to happen:

To get more precise time estimates when printing the same model over and over again.

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

[CE3_IGNITE face shield and ear saver.gcode.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/4513639/CE3_IGNITE.face.shield.and.ear.saver.gcode.txt)

The print time genius log is exactly one line long:
```
2020-04-21 14:51:21,875 Starting PrintTimeGenius
```

eyal0, 2020-04-28:
First, I'm thrilled to be included in that video.  How cool!

Now, let's see what's going wrong.  As a first step, can you go into thePTG settings page and try to select a file and analyze it?  That should start analysis immediately and you should get the gold star eventually.  Plus, you log should show a lot more.

Do that first and let me know what happens.

spierepf, 2020-04-28:
Sorry, I should have closed this. I'm guessing the PTG only analyses files as they are uploaded, and not old files that were uploaded to OctoPrint before the PTG was installed. Your plugin is fantastic. Totally deserves to be in that video. Cheers!

closed 2020-04-28 by spierepf