eyal0/OctoPrint-PrintTimeGenius #188: python3 - marlin-calc.armv71 not executable

MarekKrzyszton, 2020-04-24:
OctoPrint Version: 1.4.0 (on Octopi/rpi 3B+)
PrintTimeGenius Version (if you know): 2.2.1

What did you try: moved OctoPrint to python3

What happenned (log):
```
Exception: Running: "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/CE3_MakerBot_Headphone_Stand.gcode"
Not executable: /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l

```

chmod +x marlin-calc.armv7l 
solved the problem

just letting know
cheers

eyal0, 2020-04-28:
I just redownloaded the zip file and it looks fine to me:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/2.2.1.zip

Are you sure?

MarekKrzyszton, 2020-05-22:
Hi,
I have installed plugin through Plugin Manager.
Perhaps it's python3 related, that's why mentioned

This is what I see after ls -al (marlin-calc.armv71 has been modified by me):
[image]

eyal0, 2020-05-26:
I am using python3.  I tried uninstalling and reinstalling a few times in the normal way and everything seems fine.  Could you try to download that zip file mentioned above to your OctoPi and unzip it and see if the file permissions are correct?

marekorok, 2020-05-26:
Hi,
yes - checking the zip seems attributes is correct.

[image]

I have reinstalled once again, checked that all files are removed.
Installed from the above url, the same result... 
[image]

If no one else complains I can live with that, not a big deal.

MarekKrzyszton, 2020-05-26:
Hi,
yes - checking the zip seems attributes are correct.

[image]

I have reinstalled once again, checked that all files are removed.
Installed from the above url, the same result... 
[image]

If no one else complains I can live with that, not a big deal.

eyal0, 2020-05-27:
It's weird.  No one else has mentioned it.  I guess that you could dig into your OctoPrint and see how it is running.  Maybe in the logs there is some clue, like the user that installs isn't the same as the user that runs it?

Do you have a umask configured somewhere?

https://unix.stackexchange.com/questions/166948/why-is-my-pip-installed-python-script-not-executable-for-everyone

closed 2020-06-01 by eyal0