eyal0/OctoPrint-PrintTimeGenius #189: If Simplify3D multi-process supported?

jr-air, 2020-04-29:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.4.0

PrintTimeGenius Version (if you know): v2.2.1

What did you try: Sliced four objects and set as independent prints in one gcode (simplify3d)

What happenned: Estimated time appeared to calc only one process. 
Simplify Estimate: 15 hours
PrintGenius estimate: ~5:30 hrs
Actual Time: ~21hrs

What did you expect to happen: Full time estimate based on single gcode file

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

eyal0, 2020-04-29:
I've never heard of one of those prints.  Can you send more info, like a gcode file and some output from the PTG logs?

jr-air, 2020-04-29:
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/4555219/plugin_PrintTimeGenius_engine.log)

[Kobayashi_Fidget_Cube-v1.1-supports.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/4555230/Kobayashi_Fidget_Cube-v1.1-supports.zip)

"Multiple Process, Sequential Printing Mode"
https://www.simplify3d.com/support/articles/multi-part-printing/

In the gcode, you can search for "Process1, Process2, etc.." Same part printed over and over on different parts of the bed.

jr-air, 2020-04-29:
BTW, this plugin estimates way better than both Ocotprint "native" and Simplify3d for single prints. :)

eyal0, 2020-04-29:
First of all, I'm thrilled to hear that it's doing a good job.  I tried very hard!  PTG does an actual arlin simulation, going through all the steps of printing your gcode in super-high speed and recording how long each line took.  Thanks!

# Quick fix

Looks like the quick fix is for you to go into the PTG settings, advanced, and remove some print time history of prints that took under 2 minutes (you have 4 of them).  Then re-analyze your file and you should get a much better number.  I think that it'll be around 21 hours and 37 minutes.

# The details

I think that I see the problem in your log file.  Way at the end, if you search for the "Average scaling factor", looks like it's only 0.24!  That's pretty low, it ought to be very close to 1.  It seems that PTG computed that the total time should be 78144seconds (21.7 hours) but then scaled that by 0.24 because that's what the last five prints averaged.

Those last five prints look like this:

`18612.2490084/18748.0556116, 0.0/76.064959654, 18.499724079/77.0057332924, 0.0/28.2433313773, 0.0/57.7263332056`

Looks like maybe your most recent print took 18612 seconds compared to an estimate of 18748 (excellent, really close) and then you had some prints that took 0 seconds despite a prediction of around a minute.  So PTG took an average and came up with one-fourth.  That's lousy!

Maybe I should be giving more weight to longer prints?  So if we have:
* 10 hour estimate takes 11 hours (off by +10%), and
* 5 minute estimate takes 7 minutes (off by +40%)

Probably the correction ought to be somewhere between 10% and 40% but, like, **way** closer to 10%, right?!

That'd be pretty easy for me to fix...

PR #190 (2020-05-18): Use a weighted average for print time compensation.
This fixes #189
commit cddc3d5f (2020-05-18): Use a weighted average for print time compensation.

This fixes #189