eyal0/OctoPrint-PrintTimeGenius #192: Wrong estimated time because of wrong printer_config

jonnykl, 2020-04-30:
OctoPrint Version: 1.4.0
PrintTimeGenius Version: 2.2.1

PrintTimeGenius calculated wrong printing times (way too high or low).
After searching for the error I found that `printer_config.yaml` contained wrong values. After deleting the wrong values and deleting `print_history.yaml` (to prevent wrong compensation of the new analysis) PrintTimeGenius calculates the correct print time again.

The wrong values in `printer_config.yaml` were the following:
```
M220 S50      // set feedrate to 50%
M200 D1.75    // set filament diameter to 1.75mm
M200 D0       // set filament diameter to 0mm
```

So there is must be a bug in `printer_config.py` because the two mcodes M200 should have been merged because M and T match (not present for both mcodes) and D is present for both mcodes (see comment in `printer_config.py`).

Also I didn't find the mcode `M220 S50` in any uploaded gcode file (and it should not be in any gcode file). PrintTimeGenius listens for commands sent to the printer and always updates the printer config when it finds a matching mcode. And that is a problem. When the user manually changes the feedrate (or any other non-persistent settings) in the OctoPrint UI, PrintTimeGenius saves this new value. After powering the printer off and on it's feedrate (or the other non-persistent settings) is reset to 100% but PrintTimeGenius will calculate print times based on wrong values which leads to wrong printing times at the beginning and bad compensated printing times later.

eyal0, 2020-05-09:
Everything that you said is correct except for about M200. M200 is merged in a special way so that the final M200 D0 stays. It has to be that way so that the printer can report filament diameter without enabling volumetric mode.

Regarding the leftover settings, there's no way for PTG to know which settings come from typing in versus from the config. PTG expects that a printer will print out it's config whenever it boots. Also, printer config printing should be requests when octoprint connects. You can achieve this by adding an M503 command to octoprint's configuration.

eyal0, 2020-05-09:
Do you have a different solution that PTG could implement?

jonnykl, 2020-05-10:
> Everything that you said is correct except for about M200. M200 is merged in a special way so that the final M200 D0 stays. It has to be that way so that the printer can report filament diameter without enabling volumetric mode.

Oh okey. Then the only problematic mcodes are M220 (feed rate) and M221 (flow rate). These values aren't saved in EEPROM and thus not reported by M503. Also PTG doesn't process the received values when executing "M220" or "M221" because it's format is "FR:value" or "En Flow: value%" (instead of "M220/M221 value").

> Regarding the leftover settings, there's no way for PTG to know which settings come from typing in versus from the config. PTG expects that a printer will print out it's config whenever it boots. [...]

The user doesn't know PTG expecting to get the printer config when the printer boots. So I think this behaviour should be documented (or - if possible - let PTG execute M503, M220 and M221 after a successful connection to the printer is established).

eyal0, 2020-05-10:
The format that you mention for M220 and M221 doesn't match Marlin's gcode or even any gcode that I've heard of.

https://marlinfw.org/docs/gcode/M220.html

What kind of software do you use?

I should add a thing where the M503 is detected and forces clearing all the printer config that is stored, because we can assume that M503 will print everything out.

When M221 and/or M220 are in use, do you see them in your M503 output? Perhaps send me your M503 output?

jonnykl, 2020-05-10:
I'm using Marlin 2.0.5.3.

My output of M503 (+ M220 and M221 to set feed rate and flow):
```
Send: M220 S101
Recv: ok
[...]
Send: M221 S102
Recv: ok
[...]
Send: M503
Recv: echo:  G21    ; Units in mm (mm)
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:; Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:; Steps per unit:
Recv: echo: M92 X100.00 Y100.00 Z400.00 E100.00
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X150.00 Y130.00 Z8.00 E50.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X2000.00 Y2000.00 Z100.00 E10000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P400.00 R1000.00 T1000.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.05
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Mesh Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H215 B60 F0
Recv: echo:  M145 S1 H240 B80 F0
Recv: echo:; PID settings:
Recv: echo:  M301 P17.15 I0.80 D92.04
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X700 Y700 Z700
Recv: echo:  M906 I1 Z700
Recv: 
Recv: echo:; Driver stepping mode:
Recv: echo:  M569 S1 X Y Z
Recv: echo:  M569 S1 I1 Z
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.15
Recv: ok
Send: M220
Recv: FR:101%
Recv: ok
[...]
Send: M221
Recv: echo:E0 Flow: 102%
Recv: ok
```

As you can see the **received** values from the printer have a different format.

eyal0, 2020-05-18:
Not fixed yet, PTG needs to make sure that M503, M221, and M220 are run whenever PTG starts.

PR #193 (2020-05-18): add special handling for M220 and M221
partially fixes #192