eyal0/OctoPrint-PrintTimeGenius #195: marlin-calc should support G2 and G3

eyal0, 2020-05-21:
Off-topic for the thread, but any chance you and the guy working on the ArcWelder plugin (reducing jitter, stalls, blobs, and zits due to serial buffer inadequacies by converting straight segments into G2/G3 arcs) could get together and solve the massive time errors that occur when using PTG and ArcWelder?

_Originally posted by @KenLucke in https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/150#issuecomment-631845458_

eyal0, 2020-05-21:
@KenLucke Okay, I see the problem, marlin-calc doesn't support G2 and G3.  I think that I could fix that.

Tell me, except for getting the value wrong, does everything else work smoothly?  Is it putting a gold star on your `.aw.gcode` files after they are created?  Are the ArcWelder output files, when printing, indicating that they are getting estimates from PTG?

KenLucke, 2020-05-21:
> Tell me, except for getting the value wrong, does everything else work smoothly? Is it putting a gold star on your `.aw.gcode` files after they are created? Are the ArcWelder output files, when printing, indicating that they are getting estimates from PTG?

Yes on both.

eyal0, 2020-05-22:
@KenLucke

Great to hear, so I guess that I just need to add G02 and G03 support.

Do you have test files that I can use?  Hopefully gcode files before and after arcwelder.

From what I read, arcwelder is converting sequences of gcode that look like arcs into G02 and G03 commands, which makes the gcode files smaller, sometimes much smaller.  And that smaller size means that there is less data to send from OctoPrint to the printer, so less chance of stuttering when you print.

I also read the Marlin gcode and it seems that Marlin, internally, chops up arcs into a put of smaller linear moves and then queues those up.  So if arcwelder is doing a good job of matching the arcs and if Marlin is doing a good job of turning it back into linear motion then, between the two, the time to print should not be much affected.

@FormerLurker Is that correct?  Does arcwelder do any other stuff that I should take care of?

In my testing, I'll look at gcode before and after arcwelder and run it through marlin-calc to make sure that the timing estimates stay pretty much the same.  I'll also confirm that the new changes don't affect files without arcs at all.

So please send some test files.  Thanks!

FormerLurker, 2020-05-22:
Take a look at the marlin source, and check out g2_g3.cpp.  the interpolation code is in there.  The defaults are probably fine for doing estimates.  Some firmware may stall while printing arcs at high resolutions (segment lengths below 1mm).

I will send you files tomorrow.  I also have a command line version if you want it, and an inverse processor (convert g2/g3 back to g1/g2).

eyal0, 2020-05-22:
I read the Marlin firmware. Not only that, I also modified it to work with `marlin-calc`.

`marlin-calc`, by the way, is software to simulate gcode execution. It reads gcode and computes how long each line will take to print. I didn't have arc support before.

FormerLurker, 2020-05-22:
Neat, I will take a look at it!  I presume it is c++?

eyal0, 2020-05-22:
Sure, you can look. It's c++. It's actually the Marlin software but with most of the unnecessary parts removed and just one function called `idle` replaced. In regular Marlin, that function drives the stepper motors. In my version it just adds up how much time would be spent on acceleration, coasting, and deceleration. This allows me to have a very accurate measure of how long each line takes to execute. OctoPrint-PrintTimeGenius uses that to generate a mapping from position in file to time left printing.

I also have a version that even simulates the steppers and counts the stepper cycles but counting the clocks one-by-one is slower and doesn't provide any added accuracy.

Anyway, If you're curious, it compiles with `scons`.

eyal0, 2020-05-27:
@KenLucke and @FormerLurker I'm still waiting on gcode samples.  when I get those, I can do my testing and finally incorporate this feature into a release.

FormerLurker, 2020-05-27:
Sorry for the delay, been buried in work this past week.

Check out [this comment](https://github.com/prusa3d/Prusa-Firmware/pull/2657#issuecomment-625532750) in a pull request I created for the prusa firmware fork.  You will find a zip file called PRFiles.zip, as well as a description of the contents.  There are before/after gcode files in there, including a benchy and a single layer test print that prints multiple spirals at different speeds.  I think that should be useful for you.

If you need more gcodes, check out the ArcWelderConsole binaries [here](https://github.com/formerlurker/arcwelderlib/actions/runs/107579357).  I can't promise all of them will work (I don't have a mac to test on), but if you run the executable with the -h argument, you will get full help.  Let me know if you need any assistance with this.

eyal0, 2020-05-27:
Okay, I think that I have it working.  The time estimates between regular gcode and arcwelded gcode are similar.

Have you ever tried printing an arc-welded versus non-arc-welded to see if the print times are similar?  Were they?  In my simulation of the 3D quality benchy that you sent, I found that the arc-welded version was slower by around 240 seconds.  I did some spot checking and it looks like that slowdown is spread out throughout the file.  So I don't think that it's, like, some big jump in timing in one spot in the code.  This could be a bug in my simulation or it could be the way that the Marlin planner is doing the arc-welded code, which has many more small segments.

I'd be curious to hear if my simulation difference matches what you've seen experimentally.

FormerLurker, 2020-05-27:
The differences are firmware specific.  Unfortunately, G2/G3 support is pretty poor at the moment, which isn't entirely unexpected since they are pretty rare in 3D printing.  The difference is due to stuttering, i believe.  I've written a reverse processor that converts G2/G3 commands back to G1 using the firmware settings, and have discovered that the g1 commands almost always work without any slowdown at all.  The problem is definitely firmware related, because the slowdown still exists when printing from SD.  My pull request that you looked at only reduces the issue, and only if the firmware settings are carefully tweaked.

I'm hoping that G2/G3 support will improve as more and more people use them and report issues to tye firmware folks.  I've been trying to get to the bottom of it myself, but so far I have had only marginal success (I am not an embedded programmer at all).

Anyway, I will install your update as soon as you commit the changes, and will let you know how it goes!  To my knowledge, besides Octolapse and Octoprint, you are the first to enhance G2/G3 support due to Arc Welder, so thank you!  Let me know if I can ever be of assistance!

eyal0, 2020-05-28:
Nevermind, looks like a bug in my code.  I'm working on rooting it out.

eyal0, 2020-05-28:
Figured it out, I was calling `idle2()` too much.

Using the benchy "Quality" version that you provided, I did some tests on my new version of `marlin-calc`:

The original benchy, without arc-welding, takes ~5725 seconds.

With arc-welding, `marlin-calc` reports ~5763 seconds.  Pretty similar.

I then made a special version of `marlin-calc` that will unweld the file.  It does it printing out `G1` codes at the places where `buffer_line()` is called in `G2_G3.cpp`.  The result is a file that has no arcs and the arcs are split at the same places as what marlin will do.  The result there is again ~5763 seconds.

Seeing as the un-arc-welded and the arc-welded match very, very closely, I feel confident that `marlin-calc` is doing the right job.

eyal0, 2020-05-28:
@KenLucke all done.  Try it out?

Install this: https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/g2_g3.zip

Time to print arcwelded versus not arc-welded should be within just a couple percent of one another.

KenLucke, 2020-05-28:
> @KenLucke all done. Try it out?
> 
> Install this: https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/g2_g3.zip
> 
> Time to print arcwelded versus not arc-welded should be within just a couple percent of one another.

Will do as soon as I get a free printer in a couple of days.

FormerLurker, 2020-05-28:
@eyal0, so far so good!  I tried analyzing via the settings page, and got similar results to what you reported.  I'll let you know what happens when I run a few prints.  Will be a few days, but I'll be sure to report back.

Also, I'm quite impressed with how quickly you added arc support!  I appreciate the addition greatly.

eyal0, 2020-05-28:
Yeah, wasn't too bad.  As opposed to how ti handle G1 and G0, I tried to use the Marlin g2_g3.cpp.  Worked alright.

I also upgraded to the latest Marlin.  I think that Marlin has a bug in the G2 and G3 code.  Maybe you can look at it, too, and let me know if I'm right?  I'll explain:

G2_G3.cpp is taking a G2 or G3 command and chopping it up into little segments and then queueing those segments to eventually direct the motion of the printer.

When the printer wants to move the print head to somewhere new, it needs to know the current position and the destination but also the distance to travel.  The distance to travel affects the calculation of how fast to move in order to get there.  If it's a long distance and it's a change of direction, there needs to be a ramp down of speed before the turn and a ramp up of speed after the turn, kind of like how you need to slow down when you turn a car (this is the junction deviation or jerk parameter).

If you look at the value [here](https://github.com/MarlinFirmware/Marlin/commit/36d08f8ad354005dcb45e2907fec5ece45f309a7#diff-ba6b4689a7bef0006df8cc422ccd1df3L152), the function `planner.buffer_line()` gets the distance as an argument.  You can either set it to `0` and Marlin will compute the correct value itself [here](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/module/planner.cpp#L1888) using Pythagoras a<sup>2</sup> + b<sup>2</sup> + c<sup>2</sup> = x<sup>2</sup> or you can provide it, in case you already know the answer and you want to re-use a previous calculation.

Well, in this case it is provided, except the value it wrong.  Marlin is aiming for each segment to be of length [`MM_PER_ARC_SEGMENT`](https://github.com/MarlinFirmware/Marlin/commit/36d08f8ad354005dcb45e2907fec5ece45f309a7#diff-18dab9d337cee2f9d2209940ed08dbacR1532).  And it does that [here](https://github.com/MarlinFirmware/Marlin/commit/36d08f8ad354005dcb45e2907fec5ece45f309a7#diff-ba6b4689a7bef0006df8cc422ccd1df3R114).  But because the number of segments might not perfectly divide the length to travel, it uses the next smaller number of segments by calling `FLOOR` and makes each segment a little longer.  However, it still sends to `buffer_line` the original length!  Oh no!  It should instead be using the actual length!

To confirm that it's really a bug, I've run marlin-calc with multiple values of `MM_PER_ARC_SEGMENT`.  Here is the time to print the 3D benchy with the original code versus setting the length there to `0` so that it'll be recalculated:

| `MM_PER_ARC_SEGMENT` | Original Code          | Patched code           |
|----------------------|------------------------|------------------------|
| 1                    | 5759.21194870144699962 | 5763.28948714283615118 |
| 2                    | 5741.49710768708064279 | 5763.57120614900941291 |
| 3                    | 5733.48012508757619798 | 5763.60609510048470838 |
| 4                    | 5729.85505019807715144 | 5763.68083056076011417 |
| 5                    | 5733.92606870308918587 | 5763.88536226067753887 |

You can see how, in the unfixed code, the time to print is bouncing around, sometimes increasing and sometimes decreasing.  That wouldn't be expected.  If I change the value to 0 so that it is recomputed each time, the time to print stays far more consistent.  (It rises steadily due to the jdev/jerk requirements mentioned above.)  This seems like it's much more correct.

I'll send a patch to Marlin.

FormerLurker, 2020-05-28:
Good catch!  In fact, I already reported this exact issue.  I will post a link and you can chime in there, or post a new issue if there are some unfixed issues

PR #199 (2020-06-01): G2 ad G3 support
This fixes #195
closed 2020-06-01 by eyal0