eyal0/OctoPrint-PrintTimeGenius #197: [Feature Request] Add an option to have different levels of analyzing performance

thezak48, 2020-05-25:
Is there any chance that an option to have the ability to choose the amount of resources that are used to analyze a file, for example I have a Pi 4 but when analyzing new files the CPU barely goes above 7% utilization meaning largish files take ages to process.

eyal0, 2020-05-26:
PTG can only set the priority of marlin-calc, not the utilization.  The priority is set very low.  While marlin-calc is running, if you see only 7% in use, there are two possibilities:

1. Some other programs are using the other 93%.
2. marlin-calc is waiting on the disk.

To check the first, you could run "top" on your RPi4 while marlin-calc is running and see the percentage used by PTG versus other programs.  I'd be interested to see the results.

To check the second, you could run marlin-calc directly from the command line.  That will run at high priority.  Then you can see if the utilization is at 100% or if it's still low, maybe because it's waiting for something else.  You can also write `\time -v` before the marlin-calc command so that you get even more detailed info on the times.

To figure out what command to run, look in the PTG logs.  You'll see the previous marlin-calc invocations in there.

thezak48, 2020-05-28:
After about 6 accidental hard reboots it seems to have pulled itself together xD
Can I change my request to instead change the order of analyzing new files or allow prioritizing

eyal0, 2020-05-28:
I opened a new issue for it.

closed 2020-05-28 by eyal0