eyal0/OctoPrint-PrintTimeGenius #201: Filament length and volume always displays purge amount

Menace-Dennis, 2020-07-04:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.4.0

PrintTimeGenius Version (if you know):

What did you try: Analysing a small file

What happened: The filament length displayed the purge length of 0.03 m

What did you expect to happen: Should have displayed the amount for the something around 0.65 
If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).
[image]
[Gcode and Log.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/4872807/Gcode.and.Log.zip)

eyal0, 2020-07-06:
Weird, it works fine on my computer.  Here's what my output looks like:

```
/out/x86_64/marlin-calc.x86_64 Gcode\ File.gcode "M200 D1.75
M200 D0
M92 X80.00 Y80.00 Z400.00 E100.00
M203 X500.00 Y500.00 Z5.00 E50.00
M201 X500 Y500 Z100 E5000
M204 P500.00 R500.00 T500.00
M205 E5.00 S0.00 T0.00 X10.00 Y10.00 Z10.00
M900 K0.00"
<lots of lines...>
Processed 17792 Gcodes and 32 Mcodes. 17239 blocks
Total time: 2661.550364
Analysis: {"estimatedPrintTime": 2661.55036422873627089, "printingArea": {"maxX": 141.12199401855468750, "maxY": 200.00000000000000000, "maxZ": 7.00000000000000000, "minX": 0.10000000149011612, "minY": 20.00000000000000000, "minZ": 0.20000000298023224}, "dimensions": {"width": 141.02199401706457138, "depth": 180.00000000000000000, "height": 6.79999999701976776}, "filament": {"tool0": {"length": 675.91180419921875000, "volume": 1.62575841201246130}}}
```

Notice that the number of Gcodes, mcodes, and blocks processed are different.  Are the two of us for sure using the same file?  the height and depth are the same so I'm tempted to think that maybe it's just the version that is different.  Are you using the latest PTG version, which is 2.2.4?

Do you have some other plugins that might be modifying the gcode file such that the one that you sent me has more lines of gcode than the one which you processed?

Menace-Dennis, 2020-07-07:
Ok, I've found the issue, I started by disabling all the plugins I thought could be modifying the gcode and enabled them one by one. The culprit was the Cancel-Objects plugin. The Cancel-Objects plugin uses the object name to create a trigger of the form 

@Object 3f04c7b5-05a5-4908-a081-0ce2602ac452.stl ; (if using make from Fusion 360), or
@Object Gcode_File.stl ; if directly imported stl

which presumedly your plugin can't handle. It's interesting because it can handle smaller @ commands (like that for Octolapse) maybe it's the length of the string or the space or the dot??

[plugin_PrintTimeGenius_engine-6.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/4883853/plugin_PrintTimeGenius_engine-6.log)

eyal0, 2020-07-07:
I don't expect PTG to handle any of those! Can you explain more about how it works? Does it wrap the gcode with those tags?

Menace-Dennis, 2020-07-09:
Exactly that, it adds the names of the Meshes in as well as a NONMESH with the prefix “@object “ so that the program can jump over the code of the user wants to cancel an object