eyal0/OctoPrint-PrintTimeGenius #202: OctoPrint 1.4.1 Related Issues

jneilliii, 2020-07-06:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.4.1rc1

PrintTimeGenius Version (if you know): 2.2.4

What did you try: Clicking the file list additional information button to show additional metadata. This button has moved from the default position and files that have been analyzed by PTG break the button from working. Prior to analysis and with the plugin disabled it works as expected.

[image]

What happenned: Nothing

What did you expect to happen: The additional details to show up.

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

jneilliii, 2020-07-06:
Maybe this would be a good opportunity to remove the gold star from the file list, as requested in #183.

foosel, 2020-07-14:
FYI, turns out the breakage was caused by broken CSS, I've fixed things on OctoPrint's end as well. Being able to get rid of the star is definitely welcome though.

eyal0, 2020-07-14:
I put up a PR that should solve it. @jneilliii please test it and then I'll release.

eyal0, 2020-07-15:
> FYI, turns out the breakage was caused by broken CSS, I've fixed things on OctoPrint's end as well. Being able to get rid of the star is definitely welcome though.

Done.

PR #204 (2020-07-15): Add showStars option.
This allows users to disable the gold stars in the file list.

This fixes #202
This fixes #183
commit 3337c570 (2020-07-15): Add showStars option.

This allows users to disable the gold stars in the file list.

This fixes #202
This fixes #183