eyal0/OctoPrint-PrintTimeGenius #203: Problems with uninstall and update

hamster65, 2020-07-08:
OctoPrint Version:
 1.4.1 RC and older

PrintTimeGenius Version (if you know):
2.2.1

What did you try:
A: Start octoprint or search for updates, update print time genius
B: Tried to uninstall and reinstall print time genius

What happenned:
A: New version is found, updates seems ok, but after restart, version is still 2.2.1 and update notification pops up again

B: error, cannot uninstall:
Uninstalling plugin "PrintTimeGenius Plugin"
/home/pi/OctoPrint/venv/bin/python2 -m pip --disable-pip-version-check uninstall --yes -ctoPrint-PrintTimeGenius

Usage:
/home/pi/OctoPrint/venv/bin/python2 -m pip uninstall [options] <package> .../home/pi/OctoPrint/venv/bin/python2 -m pip uninstall [options] -r <requirements file>
...no such option: -c

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).
Odd: The installation is logged to plugin_firmwareupdater_console.log
[pluginupdate.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/4891394/pluginupdate.txt)

eyal0, 2020-07-08:
Are other plugins installing correctly?

hamster65, 2020-07-08:
yes, I have this problem for some time now and no issues with other plugins. I just removed and installed the firmware updater plugin but that did not change the log file name.

hamster65, 2020-07-08:
Correction: Same issue for DisplayLayerProgress Plugin, for both update and uninstall. I had this plugin disabled for a long time, just reenabled it and it offerend an update.

hamster65, 2020-07-08:
solved:
I went to ~/OctoPrint/venv/lib/python2.7/site-packages and removed the directories
 '~ctoPrint_PrintTimeGenius-2.2.1.dist-info'
 '~ctoprint_DisplayLayerProgress' 
'~ctoprint_PrintTimeGenius' 
'~isplayLayerProgress-1.18.1.dist-info'

That looks like a file system issue. Closed :)

closed 2020-07-08 by hamster65