eyal0/OctoPrint-PrintTimeGenius #205: Installed in OctoPrint 1.3.6 and not showing as installed

atestani, 2020-07-13:
I installed the plugin twice.  Both times it reported success.  I doesn't show up in the Plug-in Manager or can I see any difference when running a print.  Any advice?

eyal0, 2020-07-13:
It ought to show up in the plug in manager at least.  How about in the settings, do you see anything there?

Please also check the logs, which are available from the settings.  Either read them yourself or send them here and I can read them to look for errors.

atestani, 2020-07-13:
Nothing in settings.  Nothing in logs, either.  On the second install
attempt it did say that the plugin was found and a forced reinstall was
happening.  Both times it showed successful installation but there is no
trace of it.  I am doing a print now so maybe I need an OctoPrint restart
for the log to be dumped?

I keep getting messages I need to update pip which I haven't done yet.
Could this be an issue?

eyal0, 2020-07-16:
Could this be the same as #203 ?  There are instructions in that one about how to solve the issue.

Try it?

atestani, 2020-07-16:
No, not the same.  This is a MakerGear M3-ID printer I am trying to install
on and nowhere can I find any of the files referred to in #203.

There is some very wrong with your installer for us to be having these
problems.  Any fix in sight?

eyal0, 2020-07-16:
I don't know anyone else with the problem and I have hundreds of users.  Could it be a problem on your system?

Maybe you could attach your logs to a message here on GitHub and I could look at them for clues?

atestani, 2020-07-16:
Thanks...

I checked the logs and there was no mention of the plug-in anywhere.  I'll
try to install again and see if I can capture something in the logs.

atestani, 2020-07-16:
I cleaned out all of the pertinent logs and attached are two fresh ones. There *are* error messages for the plugin so hopefully you can determine the problem.  Thanks for the help.
[plugin_pluginmanager_console.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/4934925/plugin_pluginmanager_console.log)
[octoprint.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/4934926/octoprint.log)

eyal0, 2020-07-16:
https://community.octoprint.org/t/new-time-estimate-hooks-python-questions/1871

That link explains that the octoprint printtime estimation factory was added to octoprint 1.3.9. but you're on 1.3.6, right? Maybe you need a newer octoprint.

eyal0, 2020-07-16:
I should modify PTG to complain when the octoprint version is too old

eyal0, 2020-07-16:
Hmm, looks like I already have that requirement. I'm not sure why octoprint didn't catch it more gracefully.

atestani, 2020-07-16:
Yes, I am on 1.36 which is a version customized by MakerGear for their
printers.

I guess we can close-out the issue and I'll wait until MakerGear updates
their version of Octoprint which they are working on, I am told.

eyal0, 2020-07-16:
1.3.6 is 2.5 years old. I wouldn't hold my breath!

Is their code open source? I mean, it has to be by contract but is it?

Because if they didn't customize too much, I would just upgrade myself.

atestani, 2020-07-17:
I just emailed MG about this and will see what they say,  I don't use any
of the slicing or other than the basics in OctoPrint so I don't see why I
can't just upgrade.  I also asked when their new version of OctoPrint will
be released.