eyal0/OctoPrint-PrintTimeGenius #210: 'Upload to SD' fails in Python 3 Environment

b-morgan, 2020-09-12:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version:

1.4.2

PrintTimeGenius Version (if you know):

2.2.5

What did you try:

Upload to SD

What happenned:

**serial.log:**
```
2020-09-11` 19:49:30,042 - Send: M28 /test_c~1.gco
2020-09-11 19:49:30,070 - Recv: echo:Now fresh file: /test_c~1.gco
2020-09-11 19:49:30,080 - Recv: Writing to file: test_c~1.gco
2020-09-11 19:49:30,081 - Changing monitoring state from "Operational" to "Sending file to SD"
2020-09-11 19:49:30,221 - See octoprint.log for details
2020-09-11 19:49:30,258 - Changing monitoring state from "Sending file to SD" to "Offline (Error: See octoprint.log for details)"
2020-09-11 19:49:30,261 - Connection closed, closing down monitor
```
**octoprint.log:**
```
2020-09-11 19:49:30,081 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Sending file to SD"
2020-09-11 19:49:30,084 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 2185, in _monitor
    self._changeState(self.STATE_PRINTING)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 697, in _changeState
    self._callback.on_comm_state_change(newState)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/printer/standard.py", line 1116, in on_comm_state_change
    self._analysisQueue.pause() # only analyse files while idle
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/analysis.py", line 106, in pause
    q.pause()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/analysis.py", line 218, in pause
    self._do_abort()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 207, in _do_abort
    if _allow_analysis(self._plugin._printer, self._plugin._settings):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 188, in _allow_analysis
    if thermostat['target'] < 30:
TypeError: '<' not supported between instances of 'NoneType' and 'int'
2020-09-11 19:49:30,258 - octoprint.util.comm - INFO - Changing monitoring state from "Sending file to SD" to "Offline (Error: See octoprint.log for details)"
2020-09-11 19:49:30,296 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2020-09-11 19:49:30,843 - octoprint.plugins.tracking - INFO - Sent tracking event commerror_crash, payload: {'commerror_text': 'See octoprint.log for details'}
2020-09-11 19:49:30,850 - octoprint.filemanager.analysis - INFO - Analysis of entry local:Test_Cube.gcode finished, needed 0.97s
2020-09-11 19:49:46,021 - octoprint.server.api.system - INFO - Performing command for core:restart_safe: sudo service octoprint restart
```

What did you expect to happen:

File to be transferred to the printer's SD card.

**plugin_PrintTimeGenius_engine.log:**
```
2020-09-11 19:47:51,408 Starting PrintTimeGenius
2020-09-11 19:49:30,062 Not running built-in analysis.
2020-09-11 19:49:30,064 Disabled: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Test_Cube.gcode"
2020-09-11 19:49:30,072 Running: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/Test_Cube.gcode" "M200 T1 D3.00
M92 T0 E760.00
M92 T1 E760.00
M203 T0 E25.00
M203 T1 E25.00
M201 T0 E10000
M201 T1 E10000
M200 D3.00
M200 D0
M92 X100.50 Y100.50 Z1600.00 E415.00
M203 X300.00 Y300.00 Z3.00 E25.00
M201 X9000.00 Y9000.00 Z100.00 E9000.00
M204 P500.00 R3000.00 T500.00
M205 B20000.00 E10.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 J0.01
M900 K0.00"
2020-09-11 19:49:30,619 Sarge output: Junction deviation enabled
Read up all remaining commands
Processed 815 Gcodes and 55 Mcodes. 836 blocks
Total time: 203.907254
Running: "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/Test_Cube.gcode" "M200 T1 D3.00
M92 T0 E760.00
M92 T1 E760.00
M203 T0 E25.00
M203 T1 E25.00
M201 T0 E10000
M201 T1 E10000
M200 D3.00
M200 D0
M92 X100.50 Y100.50 Z1600.00 E415.00
M203 X300.00 Y300.00 Z3.00 E25.00
M201 X9000.00 Y9000.00 Z100.00 E9000.00
M204 P500.00 R3000.00 T500.00
M205 B20000.00 E10.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 J0.01
M900 K0.00"

2020-09-11 19:49:30,620 Result: {"estimatedPrintTime": 203.90725394208133, "printingArea": {"maxX": 148.24099731445312, "maxY": 148.24099731445312, "maxZ": 24.924999237060547, "minX": 0.0, "minY": 0.0, "minZ": 0.42500001192092896}, "dimensions": {"width": 148.24099731445312, "depth": 148.24099731445312, "height": 24.499999225139618}, "filament": {"tool0": {"length": 70.01959991455078, "volume": 0.4949393865724108}}, "firstFilament": 0.09267391607416701, "lastFilament": 0.9582810539523212, "progress": [[0, 203.90725394208133], [0.0233514568520842, 178.36248147626813], [0.09267391607416701, 133.67117214761095], [0.6464868255959849, 73.38464701203651], [0.9582810539523212, 15.31636382242283], [0.974661926669455, 8.440085042686547], [0.9886728007807054, 0.0], [1, 0]]}

2020-09-11 19:49:30,621 Merged result: {'analysisPending': True, 'estimatedPrintTime': 203.90725394208133, 'printingArea': {'maxX': 148.24099731445312, 'maxY': 148.24099731445312, 'maxZ': 24.924999237060547, 'minX': 0.0, 'minY': 0.0, 'minZ': 0.42500001192092896}, 'dimensions': {'width': 148.24099731445312, 'depth': 148.24099731445312, 'height': 24.499999225139618}, 'filament': {'tool0': {'length': 70.01959991455078, 'volume': 0.4949393865724108}}, 'firstFilament': 0.09267391607416701, 'lastFilament': 0.9582810539523212, 'progress': [[0, 203.90725394208133], [0.0233514568520842, 178.36248147626813], [0.09267391607416701, 133.67117214761095], [0.6464868255959849, 73.38464701203651], [0.9582810539523212, 15.31636382242283], [0.974661926669455, 8.440085042686547], [0.9886728007807054, 0.0], [1, 0]]}
2020-09-11 19:49:30,740 Disabled: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/Test_Cube.gcode" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining
2020-09-11 19:49:30,832 Recent heat-up times in seconds: 169.1834467459703, 89.74028549698414, 199.83666295500007, 208.617700643, 186.8875988200307
2020-09-11 19:49:30,836 Average heat-up: 170.85313893219706 seconds
2020-09-11 19:49:30,842 Recent cool-down times in seconds: 203.53663614799734, 197.23257772100624, 182.33358866896015, 211.71129234500086, 225.4394776490517
2020-09-11 19:49:30,843 Average cool-down: 204.05071450640327 seconds
2020-09-11 19:49:30,844 Time spent printing, actual vs predicted: 8704.217695864034/5858.111043469574, 127.76747623999836/118.35480832518815, 126.32026284304447/118.35480832518812, 3847.982181254/4262.854274912792, 623.034036481753/546.6288551356994
2020-09-11 19:49:30,848 Average scaling factor: 1.2315615844077088
2020-09-11 19:49:30,849 Compensated result: {'analysisPending': False, 'estimatedPrintTime': 520.6650887018397, 'printingArea': {'maxX': 148.24099731445312, 'maxY': 148.24099731445312, 'maxZ': 24.924999237060547, 'minX': 0.0, 'minY': 0.0, 'minZ': 0.42500001192092896}, 'dimensions': {'width': 148.24099731445312, 'depth': 148.24099731445312, 'height': 24.499999225139618}, 'filament': {'tool0': {'length': 70.01959991455078, 'volume': 0.4949393865724108}}, 'firstFilament': 0.09267391607416701, 'lastFilament': 0.9582810539523212, 'progress': [[0, 520.6650887018397], [0.09267391607416701, 349.81194976964264], [0.6464868255959849, 275.5653813552394], [0.9582810539523212, 204.05071450640327], [1, 0]], 'analysisPrintTime': 203.90725394208133, 'analysisFirstFilamentPrintTime': 70.23608179447038, 'analysisLastFilamentPrintTime': 188.5908901196585}
2020-09-11 19:49:48,265 Shutting down
2020-09-11 19:49:48,267 Saving settings to config.yaml
2020-09-11 19:49:48,268 Was saving needed? False
2020-09-11 19:51:24,879 Starting PrintTimeGenius
```

Also reported in [https://github.com/OctoPrint/OctoPrint/issues/3726#issue-699970393](url)

Grogyan, 2020-09-14:
** Firmware**

Marlin 2.0.5.3

**Setup**

SKR 1.3 + Full Graphic Full LCD

**OctoPI Version**

0.17

**OctoPrint Version:**

1.4.2

**PrintTimeGenius Version (if you know):**

2.2.5

**What did you try:**

Other plugins disabled
Upload to SD

What happened:**
Upload to SD card failed, however it worked on Python 2 environment
`2020-09-14 09:52:03,505 - octoprint.plugins.softwareupdate - INFO - Minimum free storage across all update relevant locations is 11.5GB. That is considered sufficient for updating.
2020-09-14 09:52:04,201 - octoprint.plugins.tracking - INFO - Sent tracking event printer_connected, payload: {'firmware_name': 'Marlin bugfix-2.0.x (Jun  1 2020', 'printer_port': None, 'printer_baudrate': 115200}
2020-09-14 09:52:04,378 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'base_url': 'http://192.168.20.2/', 'path': '/', 'query_string': 'l10n=en'}
2020-09-14 09:52:04,434 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2020-09-14 09:52:04,439 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.20.8
2020-09-14 09:52:04,511 - octoprint.server.util.flask - INFO - Passively logging in user OctoPortal from ::ffff:192.168.20.8
2020-09-14 09:52:04,511 - octoprint.access.users - INFO - Logged in user: OctoPortal
2020-09-14 09:52:05,245 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 8}
2020-09-14 09:52:05,874 - octoprint.server.util.sockjs - INFO - User OctoPortal logged in on the socket from client ::ffff:192.168.20.8
2020-09-14 09:52:06,406 - octoprint.plugins.tracking - INFO - Sent tracking event pong, payload: {'version': '1.4.2', 'os': 'linux', 'python': '3.7.3', 'pip': '20.2.3', 'cores': 4, 'freq': 1200.0, 'ram': 917016576, 'pi_model': 'Raspberry Pi 3 Model B Rev 1.2', 'octopi_version': '0.17.0', 'plugins': 'printtimegenius:2.2.5,navbartemp:0.14,displayz:0.2.0'}
2020-09-14 09:52:07,426 - octoprint.plugins.tracking - INFO - Sent tracking event startup, payload: {'version': '1.4.2', 'os': 'linux', 'python': '3.7.3', 'pip': '20.2.3', 'cores': 4, 'freq': 1200.0, 'ram': 917016576, 'pi_model': 'Raspberry Pi 3 Model B Rev 1.2', 'octopi_version': '0.17.0'}
2020-09-14 09:52:14,881 - octoprint.server.preemptive_cache - INFO - ... done in 10.50s
2020-09-14 09:52:50,857 - octoprint.server.util.flask - INFO - Passively logging in user OctoPortal from ::ffff:192.168.20.8
2020-09-14 09:52:50,858 - octoprint.access.users - INFO - Logged in user: OctoPortal
2020-09-14 09:52:51,109 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.20.8
2020-09-14 09:52:51,841 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.20.8
2020-09-14 09:52:51,875 - octoprint.server.util.flask - INFO - Passively logging in user OctoPortal from ::ffff:192.168.20.8
2020-09-14 09:52:51,875 - octoprint.access.users - INFO - Logged in user: OctoPortal
2020-09-14 09:52:53,445 - octoprint.server.util.sockjs - INFO - User OctoPortal logged in on the socket from client ::ffff:192.168.20.8
2020-09-14 09:52:59,346 - octoprint.filemanager.analysis - INFO - Starting analysis of local:raft-Benchy.gcode
2020-09-14 09:52:59,720 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-09-14 09:52:59,877 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Sending file to SD"
2020-09-14 09:52:59,880 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 2185, in _monitor
    self._changeState(self.STATE_PRINTING)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 697, in _changeState
    self._callback.on_comm_state_change(newState)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/printer/standard.py", line 1116, in on_comm_state_change
    self._analysisQueue.pause() # only analyse files while idle
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/analysis.py", line 106, in pause
    q.pause()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/analysis.py", line 218, in pause
    self._do_abort()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 207, in _do_abort
    if _allow_analysis(self._plugin._printer, self._plugin._settings):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 188, in _allow_analysis
    if thermostat['target'] < 30:
TypeError: '<' not supported between instances of 'NoneType' and 'int'
2020-09-14 09:52:59,921 - octoprint.util.comm - INFO - Changing monitoring state from "Sending file to SD" to "Offline (Error: See octoprint.log for details)"
2020-09-14 09:52:59,956 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2020-09-14 09:53:00,995 - octoprint.plugins.tracking - INFO - Sent tracking event commerror_crash, payload: {'commerror_text': 'See octoprint.log for details'}
2020-09-14 09:53:05,880 - octoprint.filemanager.analysis - INFO - Analysis of entry local:raft-Benchy.gcode finished, needed 6.53s
`

**Full Logs**
[PrintTimeGenius_SDCard.zip](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/5217281/PrintTimeGenius_SDCard.zip)

foosel, 2020-09-15:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/blob/master/octoprint_PrintTimeGenius/__init__.py#L188 and https://github.com/eyal0/OctoPrint-PrintTimeGenius/blob/master/octoprint_PrintTimeGenius/__init__.py#L192 assume that a target temperature has been set. That's not necessarily the case at all times, target may be `None`. That case needs to be handled gracefully here so that the attempt to compare with an integer won't cause an exception to be raised under Python 3 (Python 2 just silently ignores type compatibility errors like this and returns `False`, which is why that wasn't as easily found under Python 2).

```
Python 2.7.14 (v2.7.14:84471935ed, Sep 16 2017, 20:19:30) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> None > 30
False
```

```
Python 3.7.0 (v3.7.0:1bf9cc5093, Jun 27 2018, 04:59:51) [MSC v.1914 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> None > 30
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: '>' not supported between instances of 'NoneType' and 'int'
```

haplm, 2020-09-18:
This issue also influences other plugins, notably "Dashboard", which is stuck in "Starting", which in turns affect "OctoDash". It would really help to fix this, as right now the whole thing is kinda broken...

PR #212 (2020-09-18): Gracefully handle that target temp may be null
Fixes #210
commit 46523b62 (2020-09-18): Gracefully handle that target temp may be null

Fixes #210
closed 2020-09-18 by eyal0