eyal0/OctoPrint-PrintTimeGenius #215: Not an issue, just information after upgrading Octoprint to Python 3

TheLion, 2020-10-25:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.4.2

PrintTimeGenius Version (if you know): 2.2.6

What did you try:
I upgraded Octoprint to use Pyhton 3.

What happenned:
PTG stopped working.

What did you expect to happen:
I expected it to work ;-)

After examining the settings, I noticed that the Analysers commands (in the Advanced settings in PTG) were pointing to the Python 2 version, not Pyhton 3.

I had to change the Python 2 paths in the commands to the correct Python 3 paths.

After rebooting the Pi, everything worked just fine

eyal0, 2020-10-25:
Can you explain better the changes that are needed, so that I can make a PR?  Or can you adjust the code and make a PR?

TheLion, 2020-10-26:
Sure. Code is not my thing, but I can explain.

The analyzers commands in the advanced section of the PTG settings showed (for example the "Use Slic3r PE M73 time remaining" one)

"/home/pi/OctoPrint/venv/bin/**python2**" "/home/pi/OctoPrint/venv/lib/**python2.7**/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "{gcode}" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining

I changed that to:

"/home/pi/OctoPrint/venv/bin/**python3**" "/home/pi/OctoPrint/venv/lib/**python3.7**/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "{gcode}" --parsers slic3r_pe_print_time slic3r_pe_print_time_remaining

The other commands also showed a similar pyhton2 reference.

Maybe it is something specific to my setup, as nobody else raised an issue about this ;-)

eyal0, 2020-10-26:
Ah, yeah. When you first ran PTG is stored whichever version of python you used. So of you upgraded after you installed PTG then that's what happened.

There is no mention of python2 in all the code.

closed 2020-10-26 by eyal0