eyal0/OctoPrint-PrintTimeGenius #216: [ERROR] 'list' object has no attribute 'items'

jneilliii, 2020-11-12:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.5.0rc1

PrintTimeGenius Version (if you know): 2.2.6

What did you try: nothing specific

What happenned: error in logs

What did you expect to happen: no errors

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

Was debugging another issue and stumbled across this error in octoprint.log. Not sure if this is relevant to python 3 or octoprint 1.5.0rc1. 

```
2020-11-11 23:42:09,690 - octoprint.plugin - ERROR - Error while calling plugin PrintTimeGenius
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 549, in on_startup
    self.unmark_all_pending(dest, all_files[dest])
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 531, in unmark_all_pending
    self.unmark_all_pending(dest, v['children'])
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 527, in unmark_all_pending
    for k, v in all_files.items():
AttributeError: 'list' object has no attribute 'items'
```

This issue may be related to the issue [here](https://github.com/OctoPrint/OctoPrint/issues/3805) if you are using the files api, which is what I was running into, and will be fixed with the next 1.5.0 rc.

eyal0, 2020-11-12:
Yes, it looks like it's related.  Seems like it was a list but it's being changed back into a dict [here](https://github.com/OctoPrint/OctoPrint/commit/bed0f8d1c8ce1e3915dbffbaacf0e1c24b2d74f9).  So it should be eventually okay, right?

jneilliii, 2020-11-12:
I think so. Will report back once rc2 is out.