eyal0/OctoPrint-PrintTimeGenius #219: Widely wrong estimate at start

rodcloutier, 2020-11-22:
OctoPrint Version: 1.4.2

PrintTimeGenius Version (if you know): 2.2.6

What did you try:

Print the provided gcode file, sliced from Cura 4.8

What happened: The estimate was 18 hours

What did you expect to happen: An estimate of about 1 hour

Using Marlin 2.0.7.2
M503 Output
```
Send: M503
Recv: echo:  G21    ; Units in mm (mm)
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:; Filament settings: Disabled
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo: M92 X80.00 Y80.00 Z400.00 E93.00
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z10.00 E50.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P500.00 R1000.00 T500.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.08
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Mesh Bed Leveling:
Recv: echo:  M420 S0 Z10.00
Recv: echo:  G29 S3 I0 J0 Z0.08000
Recv: echo:  G29 S3 I1 J0 Z0.15000
Recv: echo:  G29 S3 I2 J0 Z0.12000
Recv: echo:  G29 S3 I0 J1 Z0.16000
Recv: echo:  G29 S3 I1 J1 Z0.20000
Recv: echo:  G29 S3 I2 J1 Z0.08000
Recv: echo:  G29 S3 I0 J2 Z0.15000
Recv: echo:  G29 S3 I1 J2 Z0.18000
Recv: echo:  G29 S3 I2 J2 Z0.17000
Recv: echo:  G29 S4 Z0.00000
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H200 B60 F0
Recv: echo:  M145 S1 H240 B110 F255
Recv: echo:; PID settings:
Recv: echo:  M301 P21.73 I1.54 D76.55
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X580 Y580 Z580
Recv: echo:  M906 T0 E650
Recv: 
Recv: echo:; Driver stepping mode:
Recv: echo:  M569 S1 X Y Z
Recv: echo:  M569 S1 T0 E
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.00
Recv: ok
```

eyal0, 2020-12-01:
Weird, none of your M503 settings have gotten into PrintTimeGenius.  Usually they get imported when M503 is run for the first time.

I can see it in the log:

```
Running: "/octoprint/plugins/lib/python3.8/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.x86_64" "/octoprint/uploads/gray_3.gcode"
```

On a properly working system it should look like [this](https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/221#issuecomment-736114453):

```
2020-11-30 19:21:42,117 Sarge output: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/box_insert5.gcode" "M200 D0
M220 S100
M221 T0 S125
M200 D1.75
M92 X80.00 Y80.00 Z4000.00 E102.86
M203 X250.00 Y250.00 Z2.00 E22.00
M201 X1000.00 Y1000.00 Z5.00 E1000.00
M204 P3000.00 R3000.00 T3000.00
M205 B20000.00 E5.00 S0.00 T0.00 X20.00 Y20.00 Z0.40 J0.01"
```

See all those M-codes added to the line?  You should have those, too.

PrintTimeGenius learns the M-code when you type M503 while the printer isn't printing anything.  Try to run M503 on your printer while you are not printing and then re-analyze the file.  See if the log starts to include those M-codes.  It should.

alaorath, 2020-12-05:
I'm seeing the same behavior... hilariously long print times....  I thought due to changing to spring-steel (and hence, MUCH faster bed heating times).

I'm trying adding M503 to the Octoprint "start gcode" and see if that helps (next print).

`F9_Land_Gear_Leg_x2.gcode
Uploaded: a day ago
Size: 6.4MB
Model size: 148.09mm × 152.84mm × 19.95mm
Filament: 9.95m / 70.35cm³
Estimated print time: 54:16:19
Last printed: 16 hours ago
Last print time: 06:16:56`

Snippet from log:
`2020-12-04 07:04:42,346 Not running built-in analysis.
2020-12-04 07:04:42,428 Disabled: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_g
code_comments.py" "/home/pi/.octoprint/uploads/F9_Land_Gear_Leg_x2.gcode"
2020-12-04 07:04:42,429 Running: "/home/pi/oprint/bin/python" "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_pr
ogress.py" marlin-calc "/home/pi/.octoprint/uploads/F9_Land_Gear_Leg_x2.gcode" "M220 S100
M221 T0 S100"
2020-12-04 07:04:51,357 Sarge output: ..Read up all remaining commands
Processed 219014 Gcodes and 253 Mcodes. 208122 blocks
Total time: 182416.963293
Running: "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/F9_Land_Gear_L
eg_x2.gcode" "M220 S100
M221 T0 S100"

2020-12-04 07:04:51,358 Result: {"estimatedPrintTime": 182416.96329266916   {snip}`

eyal0, 2020-12-07:
Can you provide more of the log?  I want to see if there was any compensation.  Also, I can analyze the file myself the same as you to see what happened.  But I can only do that if you send the file.