eyal0/OctoPrint-PrintTimeGenius #220: Reanalyzes the file each time

levinschmidt, 2020-11-29:
OctoPrint Version: 1.4.2
PrintTimeGenius Version (if you know): 2.2.6

Is it normal that after each start of OctoPrint all files have to be analyzed again?

eyal0, 2020-11-29:
No, that shouldn't happen. Results are stored.

eyal0, 2020-11-29:
Check the logs for errors or include the logs here.

levinschmidt, 2020-11-29:
Where is the analyzed data stored?

eyal0, 2020-11-30:
There is a data directory per plugin.  If you ssh to your printer, it might be at `~/.octoprint/data/OctoPrint-PrintTimeGenius` or something like that.  You might also have other data directories for other plugins.  All those directories need to be readable and writable by octoprint.  If there's a failure to read/write, I suspect that it might show up in the log files so you should check those.

The code was added by this commit: https://github.com/eyal0/OctoPrint-PrintTimeGenius/commit/e1e29ae0768f55d53a2479f59319391e925a4d3b

levinschmidt, 2020-12-04:
Will the data be stored in the print_history.yaml file or in another file?

 
> Check the logs for errors or include the logs here.

In the logs are no errors. The predictions for the analyzed files are always correct. But the next time you start Octoprint none of the files are analyzed.

eyal0, 2020-12-16:
Can you check if you see the file at `~/.octoprint/uploads/.metadata.json`?  That is where the progress is stored.  There should be "progress" inside for each analyzed file.

levinschmidt, 2020-12-21:
I think that will probably have been the problem. I had a script that copied files to the `~/.octoprint/uploade/` folder and apparently deleted the `~/.octoprint/uploads/.metadata.json` file.
Are there any other hidden files from PrintTimeGenius in the `~/.octoprint/uploade/` folder?
thanks for the help.

eyal0, 2020-12-21:
That file is not a PrintTimeGenius file, it is created and maintained by OctoPrint. PTG stores some added data in there.

You should not be copying files into that directory. You should be copying files into OctoPrint's "watched" folder. I can't find the official documentation on it but search the web about OctoPrint's watched folder. It's a directory next to the uploads folder and if you copy files into there, OctoPrint will treat them as if you have uploaded them through the GUI.

@foosel should consider better documenting it because my search on the web didn't turn up any official documentation about it.

closed 2020-12-21 by levinschmidt