eyal0/OctoPrint-PrintTimeGenius #221: messing up filament length

TomW1605, 2020-11-30:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

when i upload a gcode file the estimated length of filament used is calculated wrong. i have a few plugins that deal with usage calculations so i disabled them all and uploaded the same file and the calculation was correct. then i started re enabling them one by one and reuploading the same file between each one. in the screenshot below box_insert4.gcode was uploaded with print time genius disabled and box_insert5.gcode was done with it enabled. that is the only change made between the 2 files being uploaded. i have included the files so you can verify they are the same and try and reproduce the issue.

let me know if you want more info or logs

Please fill this out:

OctoPrint Version: 1.4.2 

PrintTimeGenius Version (if you know): 2.2.6 

What did you try: uploading a file and looking at the length used

What happenned: it was wrong

What did you expect to happen: it be right

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).
[image]

[box_insert4.gcode.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/5615670/box_insert4.gcode.txt)
[box_insert5.gcode.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/5615671/box_insert5.gcode.txt)

eyal0, 2020-11-30:
Can you also supply the PrintTimeGenius log?  I want to see how that file was analyzed.  You can search for "box_insert5.gcode" in the log and you should see a line that mentions marlin-calc.  It might be a few lines that also include some M-codes.

eyal0, 2020-11-30:
Also, your printer probably has some Mcodes stored in it already.  Can you type M503 to your printer and reply with the output?  I expect that the lines in the log should be a subset of the lines in the M503 output.

eyal0, 2020-11-30:
Here are my results when I run it, assuming filament width 1.75 and no volumetric extrusion (which means that your `M503` output should include a line that says `M200 D1.75` and later on another line with `M200 D0`.

```
./OctoPrint-PrintTimeGenius/octoprint_PrintTimeGenius/analyzers/marlin-calc.x86_64 ~/Downloads/box_insert5.gcode.txt "M200 D1.75
M200 D0
```

The output at the end is:

```
Analysis: {"estimatedPrintTime": 3542.22250331564146109, "printingArea": {"maxX": 193.15400695800781250, "maxY": 182.91499328613281250, "maxZ": 5.59999990463256836, "minX": 135.80499267578125000, "minY": 43.12400054931640625, "minZ": 0.00000000000000000}, "dimensions": {"width": 57.34901428222656250, "depth": 139.79099273681640625, "height": 5.59999990463256836}, "filament": {"tool0": {"length": 2079.78336629387922585, "volume": 13.26776289864279157}}}
```

So total print time should be just short of an hour, total filament length used is 2.079 meters, and volume of filament used is 5.0024 meters cubed.  This is very similar to your expectations!  So I expect that something about the M503 options and how marlin-calc was run is screwy.

TomW1605, 2020-11-30:
log extract
```
2020-11-30 19:21:40,170 Not running built-in analysis.
2020-11-30 19:21:40,179 Disabled: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_gcode_comments.py" "/home/pi/.octoprint/uploads/box_insert5.gcode"
2020-11-30 19:21:40,183 Running: "/home/pi/oprint/bin/python2" "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint/uploads/box_insert5.gcode" "M200 D0
M220 S100
M221 T0 S125
M200 D1.75
M92 X80.00 Y80.00 Z4000.00 E102.86
M203 X250.00 Y250.00 Z2.00 E22.00
M201 X1000.00 Y1000.00 Z5.00 E1000.00
M204 P3000.00 R3000.00 T3000.00
M205 B20000.00 E5.00 S0.00 T0.00 X20.00 Y20.00 Z0.40 J0.01"
2020-11-30 19:21:42,117 Sarge output: Running: "/home/pi/oprint/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint/uploads/box_insert5.gcode" "M200 D0
M220 S100
M221 T0 S125
M200 D1.75
M92 X80.00 Y80.00 Z4000.00 E102.86
M203 X250.00 Y250.00 Z2.00 E22.00
M201 X1000.00 Y1000.00 Z5.00 E1000.00
M204 P3000.00 R3000.00 T3000.00
M205 B20000.00 E5.00 S0.00 T0.00 X20.00 Y20.00 Z0.40 J0.01"
Junction deviation enabled
Junction deviation disabled
Read up all remaining commands
Processed 15270 Gcodes and 76 Mcodes. 13983 blocks
Total time: 3526.586645

2020-11-30 19:21:42,119 Result: {"printingArea": {"maxZ": 5.599999904632568, "maxX": 193.1540069580078, "maxY": 182.9149932861328, "minX": 135.80499267578125, "minY": 43.124000549316406, "minZ": 0.0}, "dimensions": {"width": 57.34901428222656, "depth": 139.7909927368164, "height": 5.599999904632568}, "firstFilament": 0.0044103488807356, "progress": [[0, 3526.586644621992], [0.00224158749255353, 3521.272774594405], [0.0044103488807356, 3503.8056730758394], [0.00935669788964515, 3442.847523054557], [0.01416904686651121, 3381.7405654821678], [0.01889400451813152, 3320.8717153726448], [0.02352720127824378, 3260.3543698599733], [0.02943048529859431, 3199.643634219005], [0.03783753078723563, 3139.6423265443673], [0.04669901116715484, 3079.622357165889], [0.05573236115339071, 3019.2585784099765], [0.06507303730007413, 2958.497520532817], [0.07425640906131521, 2898.4035433931826], [0.08292854156986863, 2838.121717405762], [0.09208715245562342, 2778.0984207184356], [0.1053473295395788, 2717.5555733784886], [0.11970281123328094, 2657.0062623240747], [0.13483753224375772, 2597.0014522746583], [0.14898327475687004, 2536.7846374689134], [0.16456806109236244, 2476.364287745073], [0.1839529135539576, 2416.1594855194908], [0.19880798232364794, 2356.08333701505], [0.2183326609056363, 2295.8317786046655], [0.23561283895089627, 2235.8004815415543], [0.2511917991980389, 2175.754956734937], [0.270342151603558, 2114.922609229711], [0.28885454733478305, 2054.494574288602], [0.303646985654714, 1994.4835036814397], [0.3219481856832618, 1934.1973461673874], [0.3404824292457983, 1874.125864595779], [0.35524136755771835, 1813.7651547880844], [0.3729832631046934, 1753.40358847306], [0.39373724632847196, 1693.373299657796], [0.40663911897891974, 1633.1137780481054], [0.4246461015458069, 1572.844160614386], [0.44363623652170875, 1511.9788564748403], [0.4565279135175445, 1451.9611664407148], [0.47465578741768827, 1391.6656424312355], [0.4942707703690973, 1331.3599751983093], [0.5096997088412347, 1271.2440409209703], [0.5245635167434497, 1211.1687149541226], [0.5443780432208364, 1151.0861565150485], [0.5630608520362907, 1090.8338253898933], [0.5754747897874497, 1030.5345119239269], [0.5957248163689778, 970.1240410203804], [0.6159238646774459, 909.774733235342], [0.6305764768769836, 849.7210292090908], [0.6506837642939436, 789.6955753613856], [0.6718936389310876, 729.4511030441481], [0.6948571661614963, 668.6736028257719], [0.7234982164887039, 608.633977531555], [0.7507031360377064, 548.4073705922015], [0.7726223369314283, 488.31971933865907], [0.7995009955328467, 428.02392901025905], [0.8245852189225523, 367.3515586603485], [0.84911305087486, 307.34605349949516], [0.876204361701043, 247.19742271017776], [0.9070942821312413, 186.9450629378557], [0.932918418741361, 126.83215235201897], [0.9592683598250425, 66.62576462423976], [0.9835704308537987, 6.963326312449226], [0.9839331048535686, 1.5130691093672795], [0.9841515831666829, 0.0], [1, 0]], "estimatedPrintTime": 3526.586644621992, "lastFilament": 0.9835704308537987, "filament": {"tool0": {"volume": 2.599728955023034, "length": 1080.84170159293}}}

2020-11-30 19:21:42,125 Merged result: {u'printingArea': {u'maxZ': 5.599999904632568, u'maxX': 193.1540069580078, u'maxY': 182.9149932861328, u'minX': 135.80499267578125, u'minY': 43.124000549316406, u'minZ': 0.0}, u'dimensions': {u'width': 57.34901428222656, u'depth': 139.7909927368164, u'height': 5.599999904632568}, 'analysisPending': True, u'firstFilament': 0.0044103488807356, u'progress': [[0, 3526.586644621992], [0.00224158749255353, 3521.272774594405], [0.0044103488807356, 3503.8056730758394], [0.00935669788964515, 3442.847523054557], [0.01416904686651121, 3381.7405654821678], [0.01889400451813152, 3320.8717153726448], [0.02352720127824378, 3260.3543698599733], [0.02943048529859431, 3199.643634219005], [0.03783753078723563, 3139.6423265443673], [0.04669901116715484, 3079.622357165889], [0.05573236115339071, 3019.2585784099765], [0.06507303730007413, 2958.497520532817], [0.07425640906131521, 2898.4035433931826], [0.08292854156986863, 2838.121717405762], [0.09208715245562342, 2778.0984207184356], [0.1 [... cut ...]
echo:  M201 X500.00 Y500.00 Z5.00 E500.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P250.00 R3000.00 T500.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; PID settings:
echo:  M301 P22.20 I1.08 D114.00
echo:; Filament load/unload lengths:
echo:  M603 L0.00 U0.00
echo:; Filament runout sensor:
echo:  M412 S1
ok
```

eyal0, 2020-12-01:
Alright, you have some unconventional settings.  Are you using Marlin firmware?

First up, you are using volumetric extrusion.  That's uncommon and I'm not sure if that's what you've intended.

You've also got`M221 T0 S125` in there so you're over-extruding by 25%.  That's kind of a lot!  Maybe you set that once and meant to unset it at some point?

If I remove the M221 and turn off volumetric extrusion, then I get the results that you expect.

Some example output here: https://github.com/MarlinFirmware/Marlin/issues/1378#issuecomment-124293594

I suggest that you wipe out the PTG config and then do another M503 and then analyze the file again.  The M221 option is missing in your M503 and it should be missing in the log, too.  Also, I expected to see M200 D0 in your M503, unless you really did intend volumetric extrusion in which case your slicer needs to be configured for that.

octoprint settings are at `~/.octoprint/config.yaml`, I think.  Or maybe they can be removed from the PTG settings page?  I don't recall.

TomW1605, 2020-12-01:
yes i am using marlin and am not intentionally using volumetric extrusion. it is turned off everywhere i can think to look (slicer, M503 and firmware). the `M200 S0` in the `M503` output should turn it off (from marlin config comments `M200 S0/S1 to disable/enable volumetric extrusion`) although i will admit to not being able to find any other reference to it. i am also not sure where the `M221` is coming from. i can't find any settings for over extrusion either.

TomW1605, 2020-12-01:
actualy the log does have `M200 D0` in it but it is on the line above `M220 S100` at the end. it appears to be missing a `\n`

eyal0, 2020-12-01:
The M200 D0 needs to come after the M200 D1.75 for volumetric to be disabled.

To tell the printer the filament width while also having volumetric be disabled, you need `M200 D1.75` followed by `M200 D0`.  The order matters.

TomW1605, 2020-12-01:
i know the order matters i was just pointing it out. i am still not sure where the M200 D1.75 is coming from in the first place. i even tried completely disabling volumetric functions in the firmware and that didn't help

eyal0, 2020-12-01:
It should be enough if you just type `M200 D1.75` into your console and then `M200 D0` in your console.  And that's it.

But I think that your `M503` output should have shown both of those and in the right order.  I don't know why you don't see both.

Look at this M503 output, for example: https://github.com/MarlinFirmware/Marlin/issues/17660#issue-605072251

See how `M200` is there 3 times?  That's what I would expect.

TomW1605, 2020-12-01:
ok i will try again tonight when the current print is done

TomW1605, 2020-12-08:
ok i think i have found the issue. Marlin does not use `M200 D0` in the `M503` report to indicate that volumetric is off. it uses `M200 S0 D1.75`. i assume (can find no reference in code) that the parser that runs to update the printer config in PTGs settings does not know about the `S` parameter and so ignores it. this would cause it to think that volumetric is on in firmware when it is not.

`M503` output code for `M200` status in Marlin code: https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/module/settings.cpp#L2968

TomW1605, 2020-12-08:
i did try running `M200` on my printer and it fixed the issue until an `M503` was run and it got overridden

TomW1605, 2020-12-08:
actually it is still a bit wonky but now estimating more filament than the slicer and octoprint (slicer 19.51m and PTG 24.41m). the logs still have `M200 D0` as the last M code so i don't think that is a volumetric thing and i will make a new issue if that is still there once we have this sorted.

eyal0, 2020-12-08:
I think that S0 option to M200 is a new thing. I'm not in front of a computer right now, just my phone, but I'll click blame on that file you linked to see when the syntax was added. I bet that it's new.

I'll need to fix marlin-calc to support this. It's been a while since I merges marlin-calc with the latest Marlin code so I'll so that and then we'll measure again and tackle whatever the next issue is.

eyal0, 2020-12-08:
Yup, as I expected.  It was added here: https://github.com/MarlinFirmware/Marlin/pull/17017/files

So I'll need a new marlin-calc.  I hope that it doesn't take too long.

eyal0, 2020-12-08:
I have a fix ready.  I also found another thing that Marlin has changed in their source code: In `M205`, the XYZE values are processed after and override the J value.  So if both XYZE and J are on the same line, XYZE should override.  In my code, it was reversed.  It only made a minor difference in my testing, however.

Would you like to try a new version with the fix?  To install it, go to the normal settings to install a plugin to OctoPrint but instead of picking PTG from the list, put this URL in:

https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/fix_M200.zip

Let me know how it goes!  Thanks.

TomW1605, 2020-12-08:
unfortunately it still appears to be thinking it is volumetric
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/5662538/plugin_PrintTimeGenius_engine.log)

also here is what it says after running `M200 S0 D1.75`:
```
2020-12-09 07:30:41,306 New printer config: M92 X80.00 Y80.00 Z4000.00 E102.86
M203 X250.00 Y250.00 Z2.00 E22.00
M201 X1000.00 Y1000.00 Z5.00 E1000.00
M204 P3000.00 R3000.00 T3000.00
M205 B20000.00 S0.00 T0.00 J0.01
M200 D1.75
```

eyal0, 2020-12-09:
Ah crap I forgot to update PTG to know about the S param in the config.  Sorry!

I uploaded a new version.  It's the same link as before so if you have the old zip, overwrite it with the new one.

Sorry again!  Thanks for testing this.

TomW1605, 2020-12-09:
it works great thanks

commit bcf24116 (2020-12-09): Merge pull request #223 from eyal0/fix_M200

Rebuild marlin-calc with M200 S param support and add support to PTG

This fixes #221
closed 2020-12-09 by eyal0