eyal0/OctoPrint-PrintTimeGenius #222: Plugin execution order

luketot, 2020-12-06:
OctoPrint Version:
1.5.1

PrintTimeGenius Version (if you know):
2.2.6

What did you try:
Upload a file.

What happenned:
PTG starts processing the file. ArcWelder also starts processing the file. Based on ArcWelder settings, a new file is created with .aw suffix and the old one is deleted. PTG now must process the new file, while any processing done on the original was a waste of time.

What did you expect to happen:
PTG should only process a file after all other processing is complete.

This has been discussed on OctoPrint git. I like the idea of a manager as they dsescribed, but perhaps you can improve PTG to attempt to only run after other plugins, while in the future, a manager would allow a user to override that default functionallity.
https://github.com/OctoPrint/OctoPrint/issues/3586

Alternatively, and specific only to ArcWelder, could you add a flag to only process files matching `*.aw.gcode`?

eyal0, 2021-01-17:
There's no way for PTG to know which other plugins might already be working on the file, so that's going to be tough.  All the plugins get notified of the upload at the same time.

I don't know a good solution for this.  I think that you might want to check with the OctoPrint repo on this one.

eyal0, 2021-02-03:
You know what, I don't think that even arcwelder is doing the right thing here.  The effort is valiant and it does good work, but rewriting a file that has been upload seems fishy to me.  Who knows how many plugins are broken in this way?

I have a solution but it would be a lot of work for ArcWelder: Rather than modify the file, store metadata about the file that indicates which changes need to be made.  Then hook the send-gcode to replace commands as needed.  This way the file wouldn't need to be modified at all.

PTG sort of does this: PTG doesn't modify the file and put timestamps in side.  Instead, PTG stores everything in metadata.

This would be a lot of work for the arcwelder plugin but it's arguably the more-correct thing to do.

closed 2021-02-03 by eyal0