eyal0/OctoPrint-PrintTimeGenius #224: Duplicating layer indicators

KenLucke, 2020-12-13:
PrintTimeGenius Version (if you know): 2.2.7

If PTG reevaluates a file it has already scanned (which happens if you use something like MultiColors which adds an M600 line at spots where you want it, then resaves a new file, or if you download an evalluated file, alter it (or not!) and upload it again using a different file name), it duplicates the M117 Layer Indicator line at each layer (throughout the whole file).

Example:
This is a file prior to upload for the first time:
```
: 
: 
;LAYER_COUNT:253
;LAYER:0
M107
G1 F300 Z0.6
:
:
```

Same file, after PTG first scan:
```
: 
: 
;LAYER_COUNT:253
;LAYER:0
M117 INDICATOR-Layer0
M107
G1 F300 Z0.6
: 
: 
```

Same file, after being altered and resaved, which caused a second scan by PTG:
```
: 
: 
;LAYER_COUNT:253
;LAYER:0
M117 INDICATOR-Layer0
M117 INDICATOR-Layer0
M107
G1 F300 Z0.6

: 
: 
```

It probably effectively doesn't cause any problems, except increasing file size slightly, but when files are on the borderline off what the gcode viewer can handle, that will make IT barf.

```
browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:83.0) Gecko/20100101 Firefox/83.0
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 3927818240
env.os.bits: 64
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.1
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.2
printer.firmware: Marlin 2.0.7.5_NIC NICW VEADBB (Nov 23 2020

```

eyal0, 2020-12-14:
PTG never rewrites files. The M117 was added by a different plugin.

I would say that this is the fault of the other plugins that are modifying uploaded files. Take PTG for example. PTG stores in it's data the ETA of a print at one minute intervals. I could have had PTG put all that info directly into the file but I didn't. PTG never modifies your gcode.

That plugin which is adding M117 could have instead been written to store the indicators in a separate file, like PTG does. And then it could have hooked the print status and injected an M117 into the command list as needed. Using this strategy, there would be no need to rewrite the files.

So I'm saying that the other plugin is misbehaving. Lots of plugins misbehave this way and it's a shame.

Do you know which plugin is doing that?

KenLucke, 2020-12-14:
> Do you know which plugin is doing that?

Apparently not :).

I'll have to do some research.  The only thing I know is operating on the file is PTG and Multi-Colors, and it happens even when I haven't used Multi-Colors (i.e., upload a file, download it and delete it from OP, then re-upload it again).  I've used MC hundreds of times and it's never done anything but add the M600 line into the file.  I'll snoop the logs and see what's happening when the file is uploaded.

Sorry for the false alarm, it just appeared it was PTG's fault.

eyal0, 2020-12-14:
Maybe it's this: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/104#issuecomment-570643663

?

eyal0, 2021-01-20:
No problem here.

I wish that pluigns would not rewrite gcode files.  OctoPrint supports storing plugin data and modifying gcode before it gets sent to a 3D printer.  I would bet that all these plugins which are rewriting gcode could pull off their functions without rewriting files.

closed 2021-01-20 by eyal0