eyal0/OctoPrint-PrintTimeGenius #225: No Information in config.yaml on multiple instances

evilmumi, 2020-12-15:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.5.2

PrintTimeGenius Version (if you know): newest

What did you try: start printing 

What happenned: get always wrong time on second instance. No information found in config.yaml

What did you expect to happen: information should stored correctly

i have to instances they created folders like /home/pi/,octoprint and .octoprint2

on normal folder everything is ok plugin works
on octoprint2 no information were stored in config.yaml so i got never estimation. Manual analyze is working.

Can you help?
need more information?
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/5697023/plugin_PrintTimeGenius_engine.log)
[plugin_PrintTimeGenius_engine2.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/5697024/plugin_PrintTimeGenius_engine2.log)

eyal0, 2020-12-15:
I don't understand.  Can you explain more?

In your second file called plugin_PrintTimeGenius_engine2.log, I see this:

```
2020-12-15 16:38:22,059 Running: "/home/pi/OctoPrint/venv/bin/python2" "/home/pi/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/analyze_progress.py" marlin-calc "/home/pi/.octoprint2/uploads/AI3M_assortment_box_2x2_v2.gcode" ""
2020-12-15 16:38:29,866 Sarge output: Running: "/home/pi/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint2/uploads/AI3M_assortment_box_2x2_v2.gcode"
```

Which has no Mcodes in it.  That's bad!  But just 1 minute later I see this:

```
2020-12-15 16:39:32,171 Sarge output: Running: "/home/pi/OctoPrint/venv/lib/python2.7/site-packages/octoprint_PrintTimeGenius/analyzers/marlin-calc.armv7l" "/home/pi/.octoprint2/uploads/AI3M_assortment_box_3x1_v3.gcode" "M220 S100
M92 X80.00 Y80.00 Z400.00 E92.60
M203 X500.00 Y500.00 Z6.00 E60.00
M201 X3000 Y2000 Z60 E10000
M204 P3000.00 R3000.00 T3000.00
M205 B20000 E5.00 S0.00 T0.00 X10.00 Y10.00 Z0.40
M200 D1.75
M200 D0"
```

Which does have mcodes.  That's good!  Maybe M503 was run between the two which is why it now has the mcodes.  Whenever M503 run, PTG learns all the Mcodes and stores them internally.  When PTG quits, then it writes those into config.yaml for use the next time.  However, it sounds like you are not seeing them being written into config.yaml.  Is that what are you seeing?

When you open config.yaml, do you see no mcodes in there in the PTG section?  Do you see *any* PTG settings in there?  Can you try to do stuff like change a setting in the PTG settings, quit OctoPrint, start it again, and see if the setting is properly saved there?  The mcodes and the rest of the settings are all saved in the same way so it should be there.

What does the PTG section of the config.yaml look like?  Maybe copy-paste it here?  Also, do settings for other plugins work fine?

Is this a new problem or have you always had this?

There are a lot of questions there!  Please answer what you can and I'll keep looking into it.  Thanks!

evilmumi, 2020-12-21:
Ok.
I have one raspberry with 2 instances of octoprint in different directories.
I test to analyze files. Thats could cause this information in logs.
Nothing from plugin printtimegenius is written in /home/pi/.octoprint2/config.yaml

after I changed one setting and restarted only one instance i got:
`plugins:
    PrintTimeGenius:
        exactDurations: false
`

from first instance

`plugins:
    PrintTimeGenius:
        print_history:
        -   analysisFirstFilamentPrintTime: 18.51590599505471
            analysisLastFilamentPrintTime: 15234.940007216208
            analysisPrintTime: 15240.647734847797
            compensatedPrintTime: 16129.549144149043
            firstFilamentPrintTime: 302.61506742239
            lastFilamentPrintTime: 16242.182231256738
            payload:
                file: /home/pi/.octoprint/uploads/switch_controller_holder_black_or_red.gcode
                filename: switch_controller_holder_black_or_red.gcode
                name: switch_controller_holder_black_or_red.gcode
                origin: local
                owner: dummy
                path: switch_controller_holder_black_or_red.gcode
                size: 15962957
                time: 16259.338053423911
            timestamp: 1574425817.531702
        -   analysisFirstFilamentPrintTime: 18.53756079921004
            analysisLastFilamentPrintTime: 26131.0194924236
            analysisPrintTime: 26136.895316416005
            compensatedPrintTime: 26476.13806235662
            firstFilamentPrintTime: 110.44893243908882
            lastFilamentPrintTime: 25995.25796984043
            payload:
                file: /home/pi/.octoprint/uploads/switch_stand_without_pin.gcode
                filename: switch_stand_without_pin.gcode
                name: switch_stand_without_pin.gcode
                origin: local
                owner: dummy
                path: switch_stand_without_pin.gcode
                size: 10999455
                time: 26053.309125348926
            timestamp: 1574385316.301367
        -   analysisFirstFilamentPrintTime: 18.537109195576022
            analysisLastFilamentPrintTime: 1286.0065088648014
            analysisPrintTime: 1292.667630880702
            compensatedPrintTime: 1436.3966981970711
            firstFilamentPrintTime: 107.96068846341223
            lastFilamentPrintTime: 1340.1676537944004
            payload:
                file: /home/pi/.octoprint/uploads/AI3M_pin.gcode
                filename: AI3M_pin.gcode
                name: AI3M_pin.gcode
                origin: local
                owner: dummy
                path: AI3M_pin.gcode
                size: 1104258
                time: 1392.5666028717533
            timestamp: 1574358500.982609
        -   analysisFirstFilamentPrintTime: 18.537472006493772
            analysisLastFilamentPrintTime: 11549.574901316259
            analysisPrintTime: 11555.317189837657
            compensatedPrintTime: 11780.407940095667
            firstFilamentPrintTime: 91.74675099272281
            lastFilamentPrintTime: 12650.190548088402
            payload:
                file: /home/pi/.octoprint/uploads/AI3M_Anycubic_i3_Mega_-_Bed_Leveling_Knob_v1.gcode
                filename: AI3M_Anycubic_i3_Mega_-_Bed_Leveling_Knob_v1.gcode
                name: AI3M_Anycubic_i3_Mega_-_Bed_Leveling_Knob_v1.gcode
                origin: local
                owner: dummy
                path: AI3M_Anycubic_i3_Mega_-_Bed_Leveling_Knob_v1.gcode
                size: 20686699
                time: 12718.164070906118
            timestamp: 1574347620.360726
        -   analysisFirstFilamentPrintTime: 18.537805082359228
            analysisLastFilamentPrintTime: 5228.943529143921
            analysisPrintTime: 5234.687977492533
            compensatedPrintTime: 5492.670953193302
            firstFilamentPrintTime: 101.89838575292379
            lastFilamentPrintTime: 5273.023211278953
            payload:
                file: /home/pi/.octoprint/uploads/AI3M_Angel_side.gcode
                filename: AI3M_Angel_side.gcode
                name: AI3M_Angel_side.gcode
                origin: local
                owner: dummy
                path: AI3M_Angel_side.gcode
                size: 1655223
                time: 5302.900075370446
            timestamp: 1574098839.82091
        printer_config:
        - M92 X80.00 Y80.00 Z400.00 E92.60
        - M203 X500.00 Y500.00 Z6.00 E60.00
        - M201 X3000 Y2000 Z60 E10000
        - M204 P3000.00 R3000.00 T3000.00
        - M205 B20000 E5.00 S0.00 T0.00 X10.00 Y10.00 Z0.40
        - M200 D1.75
        - M200 D0
`

I always had this problem.

eyal0, 2020-12-22:
Hmm.

Do you have an old version of PrintTimeGenius?

There should not be the words `print_history` in your file `~.octoprint/config.yaml`.  I removed it from there in November using this commit:

e1e29ae0768f55d53a2479f59319391e925a4d3b

When you reset the checkbox, it is normal that the plugin settings go empty.  That is expected.  The `config.yaml` file only stores values if they are different from the default.  If they are the same as the default, they are not written.

Perhaps upgrade PTG to the latest and try again?  You should not have print_history in config.yaml anymore.

evilmumi, 2021-01-13:
Thank you. Where can I find if i have the correct version?
Octoprint do not tell me to update anything.

evilmumi, 2021-06-07:
any news?
@eyal0