eyal0/OctoPrint-PrintTimeGenius #226: Install problem on Octopi

pjrestrepo, 2020-12-29:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Not found.

Please fill this out:

OctoPrint Version:
1.5.2 (Python 3)

PrintTimeGenius Version (if you know):
2.2.7

What did you try:
Installed with:  pip install "https://github.com/eyal0/OctoPrint-PrintTimeGenius/archive/master.zip" 

What happenned:
[image]

What did you expect to happen:
I expected it to install without any errors.

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).
No log. Did not install...

eyal0, 2020-12-29:
That's an uncommon way to install. Most people install from within OctoPrint's plugins menu. Can you try it that way?

It looks like you might be trying to install without first having activated the OctoPrint virtual environment.

pjrestrepo, 2020-12-29:
Thank you very much Eyal. It installed without any problem from within
OctoPrint.

Thank you for a very useful plugin, best regards, and a Happy New Year!

Pedro

eyal0, 2020-12-30:
No problem. Enjoy!

closed 2020-12-30 by eyal0