eyal0/OctoPrint-PrintTimeGenius #227: Percentage progress --> MQTT

SehlingS, 2021-01-15:
OctoPrint Version:  Version 1.5.2

PrintTimeGenius Version (if you know): Print Time Genius Plugin: 2.2.7

Hey, i use your plugin and the mqtt plugin https://github.com/OctoPrint/OctoPrint-MQTT/releases/tag/0.8.7.
The value for the left print time is correct, but the Percentage progress is not correct. It sends out the old xxxKB / xxxKB based one and not your better time based one.

I'm not sure if this is a problem of your plugin or of the mqtt one.
But i use also a api to send the data to my awtrix led display. There is also the wrong Percentage progress.

Best regards and thank you in advanced
Sascha

[image]

eyal0, 2021-01-15:
I understand.

The problem is that the file completion is sent by OctoPrint and PTG is not modifying OctoPrint, it is only modifying the way that it is displayed on the web.  Because MQTT is using the number reported by OctoPrint, I think that there is no way for MQTT to fix this.

Maybe you can fix this in your application?  You can see that in your MQTT you have 883 printTime and 1455 printTimeLeft.  883 divided by 883+1455 will give you the correct 37%.  Is it possible for your awtrix to do a little math to compute that percentage?  I'm not familiar with MQTT so I don't know myself.

eyal0, 2021-01-15:
This is the code that is computing the percentage on time instead of file progress: https://github.com/eyal0/OctoPrint-PrintTimeGenius/blob/master/octoprint_PrintTimeGenius/static/js/PrintTimeGenius.js#L35-L43

SehlingS, 2021-01-20:
Thank you very much for your feedback. I realized it via a calculation over my smarthome system loxone as you wrote in your last post.
I can't edit the app for my awtrix app, because i don't have the knowledge over java coding ;-).

closed 2021-01-20 by SehlingS