eyal0/OctoPrint-PrintTimeGenius #228: G2 Support

ereish64, 2021-01-24:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version:
1.5.3

PrintTimeGenius Version (if you know):
Latest according to octoprint

What did you try:
printing a file containing several G2 commands

What happenned:
Incredibly high estimated print time (estimated 150 hours)

What did you expect to happen:
This print is usually around 30 hours, nowhere near 150. Never had a problem with the plugin until this.

ereish64, 2021-01-24:
I would be happy to make a pull request if someone can breif me on the method this software uses to calculate print time estimates.

eyal0, 2021-01-24:
Please supply debug logs so that I can see how it happened.

This is the code that does the simulation to figure out times: https://github.com/eyal0/Marlin

Those results are adjusted as needed by the code in this project and that code is in __init__.py

If you can provide the log then it might be obvious what went wrong.

ereish64, 2021-01-24:
The log isn't looking real helpful. I got the log by going to the octoprint home page --> settings --> plugins --> PrintTimeGenius --> Advanced --> debug and clicked on the download the log link.

[plugin_PrintTimeGenius_engine.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/5862620/plugin_PrintTimeGenius_engine.txt)

eyal0, 2021-01-24:
Try to analyze a file and then look at the log again? You can force reanalysis from the settings. It should put a lot of info in the log.

Also check if logging is enabled. The OctoPrint logging settings will show it. Make sure that logging is enabled. You can turn on debug logging if you like, too. If you do that, remember to click the blue plus button.

ereish64, 2021-01-26:
Oh wow, this plugin really is incredible! The printing estimate seems much more reasonable now, it looks like it just needed some time to learn the performance of the printer.

closed 2021-01-26 by ereish64

eyal0, 2021-01-26:
Glad you like it.  You can see what kind of compensation the printer is using as it "learns" your printer by looking under the Advanced section of the PrintTimeGenius settings.