eyal0/OctoPrint-PrintTimeGenius #229: [Feature request] Show time of day when it will be finished

MagnusThome, 2021-01-28:
With longer prints especially it would be sooooo nice to have not only how many hours left but the what time of the day it will be finished. Would be great to be able to see this at a glance and not have to count in my head :-)

KevinNovak, 2021-05-18:
I would also love to have this! I used to use the [Display-ETA](https://plugins.octoprint.org/plugins/display_eta/) plugin to show and estimated time, however this was very inaccurate and I switched to PrintTimeGenius. Would be great to be able to show an ETA with PrintTimeGenius!

eyal0, 2021-05-18:
Hmm.  Have you tried using Display-ETA together with PTG?  Does it work?  What does it display?

Maybe show a screenshot with PTG and Display-ETA both running at the same time and also show the current time and we can see if the math is right.

KevinNovak, 2021-05-18:
I'll test this out and report back (currently mid-print so probably tomorrow!)

KevinNovak, 2021-05-18:
PTG seems to be working well with Display-ETA!

[image]

As you can see, if I add 2 hours and 28 mins to the current time (6:11 PM), I get 8:39 PM, which is exactly what Display-ETA is displaying!

eyal0, 2021-05-18:
That's what I would expect.  PTG doesn't modify the print time calculation only for itself, the print time calculation is modified for **all** of OctoPrint.  Display-ETA gets the estimate from the same place.

So can we consider this bug closed?

KevinNovak, 2021-05-18:
I think so. I'm not the original requester but for anyone looking for this sort of functionality they can just use Display-ETA alongside PTG. Seems to be a great combo.

eyal0, 2021-05-18:
I've on purpose not included that functionality into PTG because it's best if each plugin does just one thing, rather than try to cram everything into a single plugin.

The original requester can re-open this if needed.

closed 2021-05-18 by eyal0

MagnusThome, 2021-05-19:
Superb! Thanks a lot for finding this solution 👍