eyal0/OctoPrint-PrintTimeGenius #230: Total Time: inf

terri913, 2021-01-29:
Before reporting, check if your problem is here:
https://github.com/eyal0/OctoPrint-PrintTimeGenius/wiki/Common-problems

Please fill this out:

OctoPrint Version: 1.5.3

PrintTimeGenius Version (if you know): 2.2.7

What did you try:

What happenned: Not working
[plugin_PrintTimeGenius_engine.log](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/5896488/plugin_PrintTimeGenius_engine.log)
[bob_bot.txt](https://github.com/eyal0/OctoPrint-PrintTimeGenius/files/5896502/bob_bot.txt)

What did you expect to happen:

If relevant, upload the PrintTimeGenius log and any problematic gcode files (you might need to rename them).

eyal0, 2021-02-23:
The problem is due to the M203 line in the log.  It says that the maximum rate of extrusion is 0.  Can you type M503 on your printer and post that output.  Do you have `M205 E0` in your M503 output?

Sorry to have kept you waiting.

terri913, 2021-02-23:
Thanks so much for replying.

It is working perfectly now so I don’t know why it was having a problem before.

It is really cool of you to take the time to help me. And it’s a great plugin.

eyal0, 2021-02-27:
Glad to hear it.  Enjoy!

closed 2021-02-27 by eyal0